arduino/ArduinoCore-mbed #332: Portenta H7 - 3x I2C bus

RoboCode23, 2021-09-23:
The Portenta has 3x I2C bus and only 2 are defined in the libraries...

added and modified howmany from 2 to 3 in pins_arduino.h
``` 
#define WIRE_HOWMANY		3
#define I2C_SDA2			(PH_12)
#define I2C_SCL2			(PH_11)
```
and added into Wire.h and Wire.cpp
```
#if WIRE_HOWMANY > 2
arduino::MbedI2C Wire2(I2C_SDA2, I2C_SCL2);
#endif
```

makes all 3 work

facchinm, 2021-09-29:
Hi @RoboCode23 ,
would you mind submitting a pull request so we can properly track the authorship of this fix? Thanks a lot!

RoboCode23, 2021-09-29:
> 
> 
> Hi @RoboCode23 , would you mind submitting a pull request so we can properly track the authorship of this fix? Thanks a lot!

@facchinm Sure, first time I found out something useful, so I didn't know how this works on GitHub - pull request added, don't know if properly...

cHemingway, 2022-02-16:
I have also seen this issue, its particurally impactful on the [Portenta Machine Control](https://www.arduino.cc/pro/hardware/product/portenta-machine-control), where the front Groove connector is labelled "SCL, SDA" on the silkscreen and connected to PH_8, PH_7 (bus I2C3 on the schematic) 

These are not not listed in [pins_arduino.h](https://github.com/arduino/ArduinoCore-mbed/blob/master/variants/PORTENTA_H7_M7/pins_arduino.h) for that board, and so there is no matching Wire instance by default.

cHemingway, 2022-02-25:
@facchinm , chasing this, would it help if I submitted the same pull request as RoboCode23 so this could be merged and the "missing" I2C port added?

RoboCode23, 2022-02-25:
Hello,
yes, sorry i'm overwhelmed with other work, if somebody else submits this
that be great!
RoboCode23

Robert Kotlaba

*Virtual Garage s.r.o.**www.301.cz <http://www.301.cz> &
www.virtual-garage.eu <http://www.virtual-garage.eu> *
+420 606 171 143 <+420%20606%20171%20143>

V Sadech 1229
43401 Most
Czech Republic
IC: 06860885

cHemingway, 2022-02-27:
I have opened #413 and listed you as the author in the commit

closed 2022-03-07 by facchinm

Jenifen, 2022-09-08:
PH_8, PH_7 still not listed in [pins_arduino.h](https://github.com/arduino/ArduinoCore-mbed/blob/master/variants/PORTENTA_H7_M7/pins_arduino.h)? 
@cHemingway

pennam, 2024-11-27:
PR #413 was merged in march 2022 so this change should be included in the core releases since a long time.

commit 29b8c3a4 (2022-03-07): Add 3rd I2C bus for Arduino Portenta H7, Fixes #332
commit c1fab9ce (2022-03-07): Merge pull request #413 from cHemingway/fix_portenta_i2c_count

Add 3rd I2C bus for Arduino Portenta H7, Fixes #332