arduino/ArduinoCore-mbed #334: Serial2.begin crashes when RX Data incomming

savejeff, 2021-09-24:
I work on a project where both Serial Ports are needed. The Second HardwareSerial _UART2_ does not work correctly.

This simple example:

```

UART Serial2(8,9);

void setup() {
	Serial.begin(115200);
	Serial2.begin(9600);

	delay(3000);
	Serial.println("Start");
}

void loop() {
	if (Serial2.available()) {
		Serial.write(Serial2.read());
	}
}

```

does not reach the 	`Serial.println("Start");` line if there is Data incoming over the RX pin of the Serial2. 
if initializes fine and the forwarding works if I disconnect the RX pin during the setup and then connect. After that RX data works no problem. 

to Test this I connected a second Pico to generate some Serial Data with this sketch:
```

void setup() {

	Serial1.begin(9600);
}

void loop() {
	Serial1.println("abcdefghijklmnopqrstuvwxyz01234567890");
	delay(250);
}
```

LeoDJ, 2021-11-03:
This also happens on Serial1. I stumbled over this issue too recently.
If you don't read from the Serial buffer, the Serial1 buffer runs full after exactly 256 bytes and the RP2040 will stall.

I tried to debug into the issue, it definitely hangs at `_serial->obj->readable()` in [Serial.cpp:135](https://github.com/arduino/ArduinoCore-mbed/blob/8e4a551cd3c2d87d35f932ba144ea2c482d946d0/cores/arduino/Serial.cpp#L135) but couldn't find the implementation for `UnbufferedSerial::readable()` and thus was stumped why it would freeze.

But yes, simply dropping the incoming data if the buffer is full is the correct way imo and also how other implementations do it.

facchinm, 2021-11-03:
Hi @LeoDJ, thanks for spotting this and for the workaround!
Would you mind submitting a PR so we can give the proper attribution to the fix?

LeoDJ, 2021-11-07:
The workaround implementation was provided by @savejeff, not by me, so he should be the one to submit the PR ^^'

savejeff, 2021-11-07:
Thx @LeoDJ, but I'm not that interested in the attribution. 

I rather clearly documented the copy/paste fix. But as i stated, its just a dirty fix as receiving Data on both Serial1 and Serial2 does not work even after the fix. (At the time of testing at least)

I'm not familiar with the mbed implementation but i feel its something about there being only a single rx buffer for both serial or something like that.

closed 2021-11-08 by facchinm