arduino/ArduinoCore-mbed #336: Version 2.5.2 Portenta memory Split consecutive heap size to small

hpssjellis, 2021-09-27:
@facchinm when we use M7:M4 core split 75:25 or 100:0 the consecutive memory in the heap is too small to load a larger machine learning model. Is this something being worked on or is it a finite limit?

See edge impulse forum discussion [here](https://forum.edgeimpulse.com/t/arduino-portenta-mbed-core-version-2-5-2-released/2793/6?u=rocksetta) it is a good read.

On the positive side I got a Grayscale OLED working with a 96x96 ML model, using the 320x240 Portenta camera using the regular M7 50:50 M4 github [here](url)

[image]

The only other issue is that when I try a 320x320 camera with the above ML model it also crashes even on the M7 100:0 M4. I assume it is the camera frameBuffer same consecutive heap memory issue as the above. Any suggestions for these heap issues?

facchinm, 2021-09-28:
Hi Jeremy,
the flash split leaves the memory (RAM) map unaltered, so you won't notice any difference if you are looking for a bigger heap.
However, since the Portentat has an external 8MB RAM module (albeit slower), you could have a huge heap by just using Portenta_SDRAM library and replacing the calls to `malloc` and `free` with `ea_malloc` and `ea_free`.
Maybe it's something @janjongboom or @rajames want to investigate :wink: ?

rajames, 2021-09-28:
Hi @facchinm thanks for the suggestion. I wasn't aware of the 8MB external memory :stuck_out_tongue: 

With regards to the internal RAM. I noticed that the first 0x40000 bytes of RAM_D2 may be unused. Is that the case?

```
  ...
  RAM_D2 (xrw) : ORIGIN = 0x30000000, LENGTH = 288K
  ...
    
.lwip_sec (NOLOAD) : {
        . = ABSOLUTE(0x30040000);
        *(.RxDecripSection)
        . = ABSOLUTE(0x30040100);
        *(.TxDecripSection)
        . = ABSOLUTE(0x30040400);
        *(.RxArraySection)
        . = ABSOLUTE(0x30044000);
        *(.ethusbram)
    } >RAM_D2 AT> FLASH
```
I ran an example placing the application frame buffer in that area and it worked fine.

If it's unusued, is it possible to give this area to the application heap? Moreover is there away to use the entire RAM_D2 segment in the case when no connectivity (emac) is required?

facchinm, 2021-09-28:
@rajames you can use the RAM sections as you prefer as long as you know what you are doing :slightly_smiling_face: 

Eg. M4 uses RAM_D2 as its only memory, so if you are running a sketch on M4 the same memory can't be used by the M7 .
Same for EMAC: if you are not using the Ethernet library, that area is safe, otherwise you'd encounter awful memory corruption bugs.

On our side, we want to keep the "safe" zone as broad as possible, so it's very difficult for the end user to shoot himself in the foot. Any other "dangerous" tuning is to be considered unsupported :wink:

rajames, 2021-09-28:
@facchinm thanks for the explanation, it helps.

hpssjellis, 2021-10-02:
Linked to [Issue 38 SDRAM examples](https://github.com/arduino/ArduinoCore-mbed/issues/38)

closed 2021-11-08 by facchinm