arduino/ArduinoCore-mbed #339: Raspberry Pi Pico variant pins_arduino.h still only offers 1 x SPI, I2C, UART

Gerriko, 2021-09-29:
The Raspberry Pi Pico board's pinout tells us that there are 2 x SPI, 2 x I2C and 2 x UART. These all work fine as far as my tests have confirmed.

Please could we get the **pins_arduino.h file** updated for future releases as these dual SPI, I2C, UART ports all work fine with the ArduinoCore-mbed.

I created the following amendments (note I2C numbers reversed for backwards compatibility) and there are no clashes:

```
#pragma once
#include <macros.h>
#include <stdint.h>

#ifndef __PINS_ARDUINO__
#define __PINS_ARDUINO__

#ifdef __cplusplus
extern "C" unsigned int PINCOUNT_fn();
#endif

// Pin count
// ----
#define PINS_COUNT           (PINCOUNT_fn())
#define NUM_DIGITAL_PINS     (30u)
#define NUM_ANALOG_INPUTS    (4u)
#define NUM_ANALOG_OUTPUTS   (0u)

extern PinName digitalPinToPinName(pin_size_t P);

// LEDs
// ----
#define PIN_LED     (25u)
#define LED_BUILTIN PIN_LED

// Analog pins
// -----------
#define PIN_A0 (26u)
#define PIN_A1 (27u)
#define PIN_A2 (28u)
#define PIN_A3 (29u)

static const uint8_t A0  = PIN_A0;
static const uint8_t A1  = PIN_A1;
static const uint8_t A2  = PIN_A2;
static const uint8_t A3  = PIN_A3;

#define ADC_RESOLUTION 12

// Serial
#define PIN_SERIAL_TX (0ul)
#define PIN_SERIAL_RX (1ul)
#define PIN_SERIAL_TX1 (8ul)
#define PIN_SERIAL_RX1 (9ul)

// SPI
#define PIN_SPI_MISO  (4u)
#define PIN_SPI_MOSI  (3u)
#define PIN_SPI_SCK   (2u)
#define PIN_SPI_SS    (5u)

#define PIN_SPI_MISO1  (12u)
#define PIN_SPI_MOSI1  (11u)
#define PIN_SPI_SCK1   (10u)
#define PIN_SPI_SS1    (13u)

static const uint8_t SS   = PIN_SPI_SS;   // SPI Slave SS not used. Set here only for reference.
static const uint8_t MOSI = PIN_SPI_MOSI;
static const uint8_t MISO = PIN_SPI_MISO;
static const uint8_t SCK  = PIN_SPI_SCK;

static const uint8_t SS1   = PIN_SPI_SS1;   // SPI Slave SS not used. Set here only for reference.
static const uint8_t MOSI1 = PIN_SPI_MOSI1;
static const uint8_t MISO1 = PIN_SPI_MISO1;
static const uint8_t SCK1  = PIN_SPI_SCK1;

// Wire
#define PIN_WIRE_SDA        (6u)
#define PIN_WIRE_SCL        (7u)
#define PIN_WIRE_SDA1        (20u)
#define PIN_WIRE_SCL1        (21u)

#define SERIAL_HOWMANY		2
#define SERIAL1_TX			(digitalPinToPinName(PIN_SERIAL_TX))
#define SERIAL1_RX			(digitalPinToPinName(PIN_SERIAL_RX))
#define SERIAL2_TX			(digitalPinToPinName(PIN_SERIAL_TX1))
#define SERIAL2_RX			(digitalPinToPinName(PIN_SERIAL_RX1))

#define SERIAL_CDC			1
#define HAS_UNIQUE_ISERIAL_DESCRIPTOR
#define BOARD_VENDORID		0x2e8a
#define BOARD_PRODUCTID		0x00c0
#define BOARD_NAME			"RaspberryPi Pico"

uint8_t getUniqueSerialNumber(uint8_t* name);
void _ontouch1200bps_();

#define SPI_HOWMANY		(2)
#define SPI_MISO		(digitalPinToPinName(PIN_SPI_MISO))
#define SPI_MOSI		(digitalPinToPinName(PIN_SPI_MOSI))
#define SPI_SCK			(digitalPinToPinName(PIN_SPI_SCK))
#define SPI_MISO1		(digitalPinToPinName(PIN_SPI_MISO1))
#define SPI_MOSI1		(digitalPinToPinName(PIN_SPI_MOSI1))
#define SPI_SCK1			(digitalPinToPinName(PIN_SPI_SCK1))

#define WIRE_HOWMANY	(2)
#define I2C_SDA			(digitalPinToPinName(PIN_WIRE_SDA))
#define I2C_SCL			(digitalPinToPinName(PIN_WIRE_SCL))
#define I2C_SDA1			(digitalPinToPinName(PIN_WIRE_SDA1))
#define I2C_SCL1			(digitalPinToPinName(PIN_WIRE_SCL1))

#define digitalPinToPort(P)		(digitalPinToPinName(P)/32)

#define SERIAL_PORT_USBVIRTUAL      SerialUSB
#define SERIAL_PORT_MONITOR         SerialUSB
#define SERIAL_PORT_HARDWARE        Serial1
#define SERIAL_PORT_HARDWARE_OPEN   Serial1

#define USB_MAX_POWER	(500)

#endif //__PINS_ARDUINO__
```

alranel, 2021-09-29:
Hi @Gerriko! The RP2040 micro allows to use most pins for any purpose, so assigning SPI, I2C or Serial semantics to some specific pins in the core would be arbitrary and would restict the actual capabilities of the board which is highly customizable. For this reason we don't like the idea of having predefined combinations in the core because we couldn't have all of them. The core should instead reflect the flexibility of the board and let users choose any combination of pins.

The solution is very easy: just provide the pin numbers in your sketch when you instantiate your own instances. For example:

```
Wire myWire(6, 7);
SPI mySPI(4, 3, 2);
UART mySerial(0, 1);
```

This is simple and makes your code self-documented as you include the pin numbers explicitly.

closed 2021-09-29 by alranel

Gerriko, 2021-09-29:
Unfortunately I get an error: " 'SPI' does not name a type" using this code template:

```
#include <SPI.h>
SPI mySPI(4, 3, 2);

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

Gerriko, 2021-09-29:
@alranel I've looked at the SPI library and the solution is actually quite straightforward. I will need to use:

**arduino::MbedSPI mySPI(4, 3, 2);** or simply **MbedSPI mySPI(4, 3, 2);**

alranel, 2021-09-30:
Ouch, you're right! The correct declarations are:

```
MbedI2C myWire(6, 7);
MbedSPI mySPI(4, 3, 2);
UART mySerial(0, 1);
```