arduino/ArduinoCore-mbed #357: Calling Serial.flush() insta-kills Pico

maxgerhardt, 2021-11-02:
Consider the sketch

```cpp
void setup() {
  Serial.begin(115200);
  while(!Serial) {}
  delay(200);

  Serial.println("This is a test!");
  Serial.flush();
  Serial.println("We survived!!");
}

void loop() { Serial.println("alive.."); delay(1000); }
```

The output of which is

```
This is a test!
```

followed by a hangup of the serial monitor window (cannot be closed). The on-board LED starts blinking on the board in the pattern dictated by [`mbed_die()`](https://github.com/ARMmbed/mbed-os/blob/07e119b46754fc313947f3ce8399c68a613f0c77/platform/source/mbed_board.c#L26-L53). The hangup persists until the Pico is physically removed from the USB port.

Now consider the same sketch, but comment out the `Serial.flush();`. The output is now

```
This is a test!
We survived!!
alive..
alive..
alive..
```

Conclusion: Your `Serial.flush()` implementation kills the Pico, instantly.

(You will have to put the Pico into bootloader mode when plugging it in and selecting a fake upload port, otherwise you cannot reprogramm the Pico).

facchinm, 2021-11-03:
Should be fixed by https://github.com/arduino/ArduinoCore-mbed/pull/341 (in master but not yet released).
@maxgerhardt would you mind applying the patch manually and reporting if it fixes the issue?

maxgerhardt, 2021-11-03:
Yes, editing the file locally in `C:\Users\<user>\AppData\Local\Arduino15\packages\arduino\hardware\mbed_rp2040\2.4.1\cores\arduino\Serial.cpp` with the fix and reuploading fixes the problem.

```
This is a test!
We survived!!
alive..
alive..
```

`Serial.flush()` crashing the board in the common Serial USB configuration seems pretty crucial to me, I'm sure some sketches and libraries rely on it. The RP2040_RTC library linked above e.g. does to print its debug output. Don't you want to do a re-release with the bugfix?

facchinm, 2021-11-03:
The paln was re-releasing this week but we are waiting a bit kore to merge some more goodies, so next week the change will likely be live