arduino/ArduinoCore-mbed #360: Please clarify WiFi, BLE and Ethernet capabilities on the PortentaH7 M4 core.

hpssjellis, 2021-11-15:
As far as I can tell, WiFi, BLE and Ethernet only work when activated from the M7 core. Can someone clarify if any of these 3 can be run from the PortentaH7 M4 core, (using RPC if Serial print is needed etc) ?

Also if any of the 3 can't run from the M4 core is this a physical hardware issue or is it something that could be fixed using software?

I am also having some issues running (WiFi, BLE, Ethernet and LoRaWan) from within a thread, is that a known issue?

My reasoning for the question is: I do a lot of CPU intensive [EdgeImpulse.com ](https://www.edgeimpulse.com/) machine learning vision coding with the PortentaH7 connected to either the Ethernet or LoRa Vision Shields and the machine learning code must be on the faster M7 core, meaning communication needs to be done from the slower M4 core. The only communication success I have had using the M7 Core is with bare bones LoRa peer-to-peer communication directly using the Murata module (see [MKRWAN issue 107](https://github.com/arduino-libraries/MKRWAN/issues/107)), even Arduino LoRaWan using the [MKRWAN library](https://github.com/arduino-libraries/MKRWAN) seems to have some threading issues. 

Sorry to be a pain, but the whole point of the dual core Arduino is to be able to put extra processes on the other core.

Jeremy aka rocksetta on Twitter

hpssjellis, 2021-12-09:
@facchinm does WiFi, BLE and Ethernet only work when activated from the M7 core?

Is it a hardwired issue or a software issue?

hpssjellis, 2023-01-04:
These capabilities seem to be running fine now

hpssjellis, 2023-10-26:
> Hi, how did you solve the problem? I would like to run an access point and a simple webserver with M4 Can you give me some pointers?

Hmmm, not sure what I did to close the issue. Sorry that was a while ago. Do you jave my library? The "portenta pro community solutions"? I may have put an example there?

MMirabel, 2023-10-26:
Yes I'm looking at the examples you developed in "portenta pro community solutions" but I haven't found the solution to my problem yet.
It seems that with M4 you can only control the led.

hpssjellis, 2023-10-26:
@facchinm @MMirabel 

So I tested wifi on the M4 PortentaH7 core after making sure the code worked on the M7 core using [dot181-wifi-webserver](https://github.com/hpssjellis/portenta-pro-community-solutions/tree/main/examples/dot1-portentaH7-examples/dot18-wifi-webserver/dot181-wifi-webserver). 

I did run the STMH747_System-->wifiFirmwareUpdater but not sure if I had to.

Then cleaned the M7 core and setup the M4 core using  [dot122-dual-easy](https://github.com/hpssjellis/portenta-pro-community-solutions/blob/main/examples/dot1-portentaH7-examples/dot12-utilities/dot122-dual-easy/dot122-dual-easy.ino)

and  basically copied the first example into the M4 core. It compiled and ran but no WiFi connectivity.

I then cleaned the M4 core and tried to run the WiFi code on the M7 core and caused the [Orange Led of Hell](https://github.com/arduino/ArduinoCore-mbed/issues/608)    which I am excited that I now know the steps I took, but that is my 9th Portenta I have fried and it is getting a bit expensive.

Unfortunately @MMirabel my suggestion is to stay away from inner M4 core IOT. If anyone has had better luck can they please post here.

hpssjellis, 2023-10-27:
Good detective work @MMirabel for your serial solution reminder the Portenta has several serial ports. Some of them are hard to find.  This  examples touches on some of them    [dot42-UART1-M4-to-M7.ino](https://github.com/hpssjellis/portenta-pro-community-solutions/blob/main/examples/dot4-portenta-breakout-board/dot42-UART1-M4-to-M7/dot42-UART1-M4-to-M7.ino)

razvanphp, 2024-09-20:
Duplicate of #790

See my https://github.com/arduino/ArduinoCore-mbed/issues/790#issuecomment-2362681070, this is now mentioned in the [documentation](https://docs.arduino.cc/tutorials/giga-r1-wifi/giga-dual-core/#m4-support).

We should close thos.