arduino/ArduinoCore-mbed #365: Clarification on Blocked RADIO Peripheral IRQ

ElijahBasile, 2021-11-29:
Hello,

I am seeking clarification on why access to the RADIO IRQ Handler weak link directly is blocked? I wish to use the RADIO peripheral in the NRF52840 on the Arduino Nano 33 BLE directly, however this blockage prevents custom protocols, and support for the NRF/IEEE as well as the BLE radio modes available on the device. 

https://github.com/arduino/ArduinoCore-mbed/blob/64bf2aa92bf7fe26ab08aae5f4340f2c2920ebaf/libraries/mbed-memory-status/startup_example.S.txt#L240

facchinm, 2021-11-29:
Hi @ElijahBasile ,
we indeed link the `RADIO_IRQHandler` function non-weak here https://github.com/ARMmbed/mbed-os/blob/master/connectivity/drivers/ble/FEATURE_BLE/TARGET_NORDIC/TARGET_NORDIC_CORDIO/TARGET_NRF5x/stack/sources/pal_bb.c , but it should be straightforward to use the `PalBb*` APIs to redirect it to the handler you need.
For example, a custom protocol could be activated this way

```
#include "pal_bb.h"
PalBbRegisterProtIrq(BB_PROT_15P4, nullptr, myRadioCallback);
PalBbSetProtId(BB_PROT_15P4);
```

I didn't test it in real time but it compiles just fine and does exactly what you need :slightly_smiling_face:

closed 2021-11-29 by facchinm

ElijahBasile, 2021-12-20:
Tested in real-time. This solution works great. Also, for others, another route can be utilizing the NVIC_SetVector() function (documentation below) to have custom IRQ Handler.

https://www.keil.com/pack/doc/CMSIS/Core/html/group__NVIC__gr.html#gab43c1c59d5c081f1bc725237f4b1f916