arduino/ArduinoCore-mbed #366: va_start, va_end not defined

fjansson, 2021-11-29:
I'm trying to use the Radio library on the Raspberry Pico with ArduinoCore-mbed. It does not compile, complaining about va_start not being defined. 

```
$USER/Documents/Arduino/libraries/Radio/src/SI4705.cpp: In member function 'void SI4705::_sendCommand(int, int, ...)':
$USER/Documents/Arduino/libraries/Radio/src/SI4705.cpp:530:5: error: 'va_start' was not declared in this scope
     va_start(params, cmd);
     ^~~~~~~~
```
(On OSX)

to reproduce:
* In the Arduino library manager, search for "Radio", install the Radio library by Matthias Hertel. I have version 2.0.0, the latest.
Documentation: http://www.mathertel.de/Arduino/RadioLibrary.aspx
* Open an example: Examples->Radio->TestRDA5807M
* Select the board "Raspberry Pi Pico"
* try to compile
(There's one error before about SDA not being defined, seems unrelated to this)

I tried adding includes to my main program, `#include<cstdarg>` or `#include<stdarg.h>` where va_start etc are supposed to be defined, this did not help.

I looked at cstdarg and stdarg.h, they are shipped with the *compiler*.  For me, on Linux this time, I found these files:
`.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include/c++/7.2.1/cstdarg`
`.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/lib/gcc/arm-none-eabi/7.2.1/include/stdarg.h`
Both of them contain definitions of va_start etc, but inside some complicated `#ifdef` which I didn't follow.

I believe this to be a bug in ArduinoCore-mbed (and not in the Radio library) since after `#include<cstdarg>` , `va_start` etc are supposed to work.

fjansson, 2021-11-29:
Here is a stand-alone test case - does not require the Radio library. This compiles when Board is set to Arduino Uno, but not Raspberry Pi Pico.

```
void setup() {
}
void loop() {
}
// simplified from the Radio library
void sendCommand(int cnt, int cmd, ...)                                                                                                     
{                                                                                                                                                    
  uint8_t cmdData[12];                                                                                                                               
                                                                                                                                                     
  va_list params;                                                                                                                                    
  va_start(params, cmd);
}    
```

Compiler output:
```
$USER/Documents/Arduino/va_start_test/va_start_test.ino: In function 'void sendCommand(int, int, ...)':
va_start_test:16:3: error: 'va_start' was not declared in this scope
   va_start(params, cmd);
   ^~~~~~~~
$USER/Documents/Arduino/va_start_test/va_start_test.ino:16:3: note: suggested alternative: 'stat'
   va_start(params, cmd);
   ^~~~~~~~
   stat
exit status 1
'va_start' was not declared in this scope
```

facchinm, 2021-11-30:
Hi @fjansson ,
I got your second example compiling just fine in Linux after adding 
```
#include <stdarg.h>
```
on top. I now asked some colleagues with OSX to test it out but I think it should work there too.
Which version of the core are you targeting? I tested with `mbed_rp2040` version 2.6.1 
Thanks!

fjansson, 2021-11-30:
Hi @facchinm ,
thank you for the reply! Adding `#include <stdarg.h>` in my second example makes it compile for me too on OSX. 
I used "Arduino MBED OS RP2040 Boards" in the Boards manager, version 2.5.2. I updated to 2.6.1 just now, it behaves the same.

I'll see if I can fix the radio library with the same include. I realized just now that just putting it in my main file is not enough, it should be in the library's .c or .cpp which actually calls va_start.

Still the other Arduino flavors, e.g. Uno, behave differently because my test script and the radio library compile there without the include line - maybe they already include stdarg.h by default?

facchinm, 2021-11-30:
>  maybe they already include stdarg.h by default?

Very likely, either directly or as an indirect dependency.
I'm closing the issue since there's a confirmed fix :slightly_smiling_face: