arduino/ArduinoCore-mbed #367: PIO with dma

pschatzmann, 2021-11-30:
I would like to use the pio together with dma access but the following methods are missing:

The linker gives: undefined reference to ...
- dma_irqn_get_channel_status
- dma_irqn_set_channel_enabled
- dma_irqn_acknowledge_channel

Is there a work around ?

pschatzmann, 2021-12-31:
These methods should be part of the hardware/dma.h.

It seems that your core is based on a very old pico-sdk version. Is there any chance that you plan to make a new version available soon?

My [I2S library](https://github.com/pschatzmann/rp2040-i2s) is compiling now but I needed to add the missing methods in my code which is not really a good solution.

Gerriko, 2022-05-04:
I see this issue has not been updated/resolved, so I thought to comment as I can now see the related **hardware/dma.h** file inside the latest ArduinoCore-mbed version for the Rasberry Pico board. 

Doubt if my code example below is directly related to the above issue, but just wanted to add that I could get the standard pico-examples "hello_dma" from Raspberry Pi Pico to work with just one mod to get the output displayed on the Serial monitor.

```
#include <stdio.h>
#include "hardware/dma.h"

// Data will be copied from src to dst
const char src[] = "Hello, world! (from DMA)";
char dst[count_of(src)];

void setup() {
  // put your setup code here, to run once:
    Serial.begin(115200);
    while (!Serial) {;;}
    
    // Get a free channel, panic() if there are none
    int chan = dma_claim_unused_channel(true);

    // 8 bit transfers. Both read and write address increment after each
    // transfer (each pointing to a location in src or dst respectively).
    // No DREQ is selected, so the DMA transfers as fast as it can.

    dma_channel_config c = dma_channel_get_default_config(chan);
    channel_config_set_transfer_data_size(&c, DMA_SIZE_8);
    channel_config_set_read_increment(&c, true);
    channel_config_set_write_increment(&c, true);

    dma_channel_configure(
        chan,          // Channel to be configured
        &c,            // The configuration we just created
        dst,           // The initial write address
        src,           // The initial read address
        count_of(src), // Number of transfers; in this case each is 1 byte.
        true           // Start immediately.
    );

    // We could choose to go and do something else whilst the DMA is doing its
    // thing. In this case the processor has nothing else to do, so we just
    // wait for the DMA to finish.
    dma_channel_wait_for_finish_blocking(chan);

    // The DMA has now copied our text from the transmit buffer (src) to the
    // receive buffer (dst), so we can print it out from there.
    Serial.println(dst);

}

void loop() {
  // put your main code here, to run repeatedly:
}
```

pschatzmann, 2022-05-07:
Cool. My https://github.com/pschatzmann/rp2040-i2s library is compiling now w/o any issues. 
Sometimes it's worth just to wait...

closed 2022-05-07 by pschatzmann