arduino/ArduinoCore-mbed #372: HardFault when calling nicla::begin in Platformio.

gutzchi, 2021-12-13:
https://github.com/arduino/ArduinoCore-mbed/blob/838c70238aaee11c7a8a32cf570de4f51ce240f7/libraries/Nicla_System/src/Nicla_System.cpp#L25-L35

Good morning. 

I have tried uploading the nicla-blink code using the Platformio IDE. With the Arduino IDE, the code works fine. However, with Platformio, I get a hard fault from the mbed OS:

```
++ MbedOS Fault Handler ++

FaultType: HardFault

Context:
R   0: 00000000
R   1: 2000345C
R   2: 00000001
R   3: 20000BF5
R   4: 00002000
R   5: 00000000
R   6: 20000BEC
R   7: 00000000
R   8: 00000000
R   9: 00000000
R  10: 00000000
R  11: 00000000
R  12: 00016E79
SP   : 20003408
LR   : 00010753
PC   : 00010764
xPSR : 41000000
PSP  : 200033E8
MSP  : 2000FFC0
CPUID: 410FC241
HFSR : 40000000
MMFSR: 00000000
BFSR : 00000000
UFSR : 00000001
DFSR : 00000000
AFSR : 00000000
Mode : Thread
Priv : Privileged
Stack: PSP

-- MbedOS Fault Handler --

++ MbedOS Error Info ++
Error Status: 0x80FF013D Code: 317 Module: 255
Error Message: Fault exception
Location: 0x10764
Error Value: 0x200087B4
Current Thread: main Id: 0x2000345C Entry: 0x17EE3 StackSize: 0xC00 StackMem: 0x20002838 SP: 0x20003408
For more info, visit: https://mbed.com/s/error?error=0x80FF013D&tgt=NICLA
-- MbedOS Error Info --
```
Using the debugger, the fault is triggered by the `th.start(&nicla::pingI2CThd);` call inside the `nicla::begin()` function. The code used is the following one:

```
#include "Nicla_System.h"

void setup() {
  Serial.begin(115200);
  while(!Serial);
  
  nicla::begin();
  nicla::leds.begin();
}

void loop() {
  nicla::leds.setColor(green);
  delay(1000);
  nicla::leds.setColor(off);
  delay(1000);
} 
```
The platformio.ini file is the following one:
```
[env:nicla_sense_me]
platform = nordicnrf52
board = nicla_sense_me
framework = arduino
monitor_speed=115200
debug_tool = cmsis-dap
upload_protocol = cmsis-dap
```
I have previously posted the issue in the Platformio community and they have told me that I should post it here(https://community.platformio.org/t/arduino-mbed-thread-issue-on-arduino-nicla/24915).

Thank you for your attention.

facchinm, 2021-12-13:
Hi @gutzchi ,
the only thing that comes to my mind is that platformio could be using a different version of the compiler. Any other explanation, given the exact same source code and compilation flags is very unlikely.
Could you paste here the verbose output from the platformio build?

gutzchi, 2021-12-14:
Thank you for your quick response, @facchinm.
Here you have the build log. 

[compiler_log.txt](https://github.com/arduino/ArduinoCore-mbed/files/7712526/compiler_log.txt)

facchinm, 2021-12-14:
It would be interesting to see the version of `arm-none-eabi-g++` ; I think pio is using either the system-wide compiler or their own, so the initial idea seems to be confirmed... 
@gutzchi since you already opened the discussion on their channel, would you mind asking them for some additional info? Thx

facchinm, 2021-12-14:
Sorry, I found the info I needed 
```
toolchain-gccarmnoneeabi 1.80201.190214 (8.2.1)
```
So the toolchain is definitely different from the one we are using, but one other issue could be the cohexistence of `Arduino_BHY2` and `Nicla_System` libraries https://github.com/arduino/ArduinoCore-mbed/blob/838c70238aaee11c7a8a32cf570de4f51ce240f7/libraries/Nicla_System/src/Nicla_System.cpp#L3-L9

Can you share the exact sketch you are compiling? The default Blink_Nicla should not pull in `Arduino_BHY2` lib

maxgerhardt, 2021-12-14:
Your compiler log says

```
Dependency Graph
|-- <Arduino_BHY2> 1.0.3 (C:\Users\gutzchi\Box\Tecnodiabetes\Arduino Nicla\PlatformioSketches\mBedOSFaultTest\.pio\libdeps\nicla_sense_me\Arduino_BHY2)
|   |-- <Nicla_System> 1.0 (C:\Users\gutzchi\.platformio\packages\framework-arduino-mbed\libraries\Nicla_System)
|   |   |-- <Wire> (C:\Users\gutzchi\.platformio\packages\framework-arduino-mbed\libraries\Wire)
|   |-- <Wire> (C:\Users\gutzchi\.platformio\packages\framework-arduino-mbed\libraries\Wire)
|   |-- <ArduinoBLE> 1.2.1 (C:\Users\gutzchi\Box\Tecnodiabetes\Arduino Nicla\PlatformioSketches\mBedOSFaultTest\.pio\libdeps\nicla_sense_me\ArduinoBLE)
|-- <ArduinoBLE> 1.2.1 (C:\Users\gutzchi\Box\Tecnodiabetes\Arduino Nicla\PlatformioSketches\mBedOSFaultTest\.pio\libdeps\nicla_sense_me\ArduinoBLE)
|-- <Nicla_System> 1.0 (C:\Users\gutzchi\.platformio\packages\framework-arduino-mbed\libraries\Nicla_System)
|   |-- <Wire> (C:\Users\gutzchi\.platformio\packages\framework-arduino-mbed\libraries\Wire)
```
which doesn't match the seen libraries when I use your above `platformio.ini`. I get with that only 

```
Dependency Graph
|-- <Nicla_System>
|   |-- <Wire>
```

I.e., it does not by default pull in the `Arduino_BHY2` library. Did you use a different sketch than the provided one to generate the compiler log? 

In any case, if PlatformIO generates code that crashes when the `Arduino_BHY2` library is used and Arduino-IDE does not, that's a fault in PlatformIO, not in this Arduino core, and possibly related to the different compiler version. The board is contained in the nordicnrf52 platform where in issue about that should be opened https://github.com/platformio/platform-nordicnrf52/issues.

gutzchi, 2021-12-14:
I'm very sorry, at some point I included the BHY2 library (and the ArduinoBLE), or I copied the `platformio.ini` file of another project. I'm very sorry about this mistake, it seemed fishy to me but I thought that maybe the library was automatically included, my bad. 

Here is the build log I should have attached, the platformio configuration file and the sketch are the ones from above, for sure.
[build_log_nicla.txt](https://github.com/arduino/ArduinoCore-mbed/files/7714798/build_log_nicla.txt)

Actually, if I upload the sketch, then I don't get the hard fault, but it gets stuck in the same call.

If, on the other hand, I include the `Arduino_BHY2` library, I get the hard fault. If I understand properly, from the code below:

https://github.com/arduino/ArduinoCore-mbed/blob/838c70238aaee11c7a8a32cf570de4f51ce240f7/libraries/Nicla_System/src/Nicla_System.cpp#L3-L9

the preprocessor should define the macro `NO_NEED_FOR_WATCHDOG_THREAD`. However in the debugger the following lines are executed (which trigger the fault):

https://github.com/arduino/ArduinoCore-mbed/blob/838c70238aaee11c7a8a32cf570de4f51ce240f7/libraries/Nicla_System/src/Nicla_System.cpp#L30-L33

How can this be?

@maxgerhardt I agree with you, but I have posted the issue here because I was told to. Once @facchinm answers me, I will post it on the link you have provided me. Thank you very much to both of you.