arduino/ArduinoCore-mbed #376: BLE pairing protocol allways ends up with SEC_STATUS_UNSPECIFIED

afpineda, 2021-12-26:
No matter how you configure the SecurityManager, the pairing protocol ends up with **"SEC_STATUS_UNSPECIFIED"** (0x88) at the "pairingResult" event, in the peripheral's side (Mbed API). On the central's side, it ends up with a timeout.

Board: Arduino Nano 33 BLE.
Tested both in Android (nRF app) and Windows 10 (with BT 5.0 radio).

Pairing is mandatory in Windows 10 API in order to retrieve services and characteristics, so this is a blocking issue.

This is the involved code (not the full code):
```c++
        // Stack initialization
        BLE& ble = BLE::Instance();
        ble_error_t error = ble.init();
        ERRORCHECK(error,nullptr);
        while (!ble.hasInitialized()) 
            ble.processEvents();
        error = ble.securityManager().init(
            /* enableBonding */ true,
            /* requireMITM */ false,
            /* iocaps */ SecurityManager::IO_CAPS_NONE,
            /* passkey */ nullptr,
            /* signing */ false,
            /* dbFilepath */ nullptr);
        // Security Manager initialization
        error = ble.securityManager().setPairingRequestAuthorisation(true);
        ERRORCHECK(error,nullptr);
        error = ble.securityManager().allowLegacyPairing(true);
        ERRORCHECK(error,nullptr);
  
        // GAP initialization
        error = ble.gap().enablePrivacy(false);
        ERRORCHECK(error,nullptr);
        error = DeviceInformationService::add_service(ble,
            BLE_MANUFACTURER,
            BLE_DEVICE_NAME,
            nullptr, // Serial number
            BLE_HW_REV,
            BLE_FW_REV
        );
       ERRORCHECK(error,nullptr);
       ...
```
and the event handler:

```c++
void pairingRequest (ble::connection_handle_t connectionHandle)
{
    Serial.println("PAIRING REQUEST");
    BLE& ble = BLE::Instance();
    ble_error_t error;
    error = ble.securityManager().acceptPairingRequest(connectionHandle);
    if (error) {
        Serial.print("acceptPairingRequest failed with code ");
        Serial.println(error);
    }
}

void pairingResult (
    ble::connection_handle_t connectionHandle, 
    ble::SecurityManager::SecurityCompletionStatus_t result)
{
    Serial.println("PAIRING RESULT");
    Serial.println(result);
    if (result== ble::SecurityManager::SecurityCompletionStatus_t::SEC_STATUS_SUCCESS) {

    } else {
        start_advertising();
    }
}
```

haroonalisa, 2022-07-27:
Can anyone mention if this has been fixed or is there a workaround?

ufocia, 2023-01-25:
I'm having a similar problem. Looks similar to [this](https://github.com/ARMmbed/mbed-os/issues/15326).