arduino/ArduinoCore-mbed #377: How can I make the external QSPI flash work with Arduino NANO 33 BLE?

bigbearishappy, 2022-01-03:
Hello, I am working with my arduino 33 ble.
And I want to make it work fine with a QSPI flash.
I find a [demo code](https://github.com/arduino/ArduinoCore-mbed/blob/master/libraries/USBMSD/examples/Nano33BLE_FlashMassStorage/Nano33BLE_FlashMassStorage.ino) in this project. 
But it's based on the nrf52840's internal flash.

What should I do if I want to make the external QSPI flash works with nano 33 BLE?

facchinm, 2022-01-10:
Hi @bigbearishappy ,
to use mbed classes you first need to modify nano33 config file adding `"target.components_add" : ["QSPIF"],` (like https://github.com/arduino/ArduinoCore-mbed/blob/master/variants/NICLA/conf/mbed_app.json#L20 ) and recompile `libmbed` using `mbed-os-to-arduino` script (https://github.com/arduino/ArduinoCore-mbed#adding-an-mbed-target) .

Once done, you can replace this line https://github.com/arduino/ArduinoCore-mbed/blob/master/libraries/USBMSD/src/Singleton.cpp#L82 with an object of type `QSPIFBlockDevice`, with the correct pins as you wired them.

Let me know if you need any help with these steps

facchinm, 2022-01-10:
You procedure was correct but I forgot one modifications :slightly_smiling_face: 

Here's the full patch for mbed_app.json

```patch
+++ b/variants/ARDUINO_NANO33BLE/conf/mbed_app.json
@@ -1,6 +1,12 @@
 {
   "macros": [
-    "NRFX_GPIOTE_CONFIG_NUM_OF_LOW_POWER_EVENTS=8"
+    "NRFX_GPIOTE_CONFIG_NUM_OF_LOW_POWER_EVENTS=8",
+    "QSPI_FLASH1_IO0=p1",
+    "QSPI_FLASH1_IO1=p1",
+    "QSPI_FLASH1_IO2=p1",
+    "QSPI_FLASH1_IO3=p1",
+    "QSPI_FLASH1_CSN=p1",
+    "QSPI_FLASH1_SCK=p1"
   ],
   "target_overrides": {
     "*": {
@@ -12,7 +18,9 @@
       "rtos.main-thread-stack-size": 32768,
       "cordio.max-connections": 5,
       "platform.all-stats-enabled": true,
-      "target.mbed_app_start": "0x10000"
+      "target.mbed_app_start": "0x10000",
+      "target.components_add" : ["QSPIF"],
+      "target.device_has_add" : ["QSPI"]
     }
   }
```

Of course replace `p1` with the pins you are actually using to connect the external QSPI flash

bigbearishappy, 2022-01-12:
> You procedure was correct but I forgot one modification 🙂
> 
> Here's the full patch for mbed_app.json
> 
> ```diff
> +++ b/variants/ARDUINO_NANO33BLE/conf/mbed_app.json
> @@ -1,6 +1,12 @@
>  {
>    "macros": [
> -    "NRFX_GPIOTE_CONFIG_NUM_OF_LOW_POWER_EVENTS=8"
> +    "NRFX_GPIOTE_CONFIG_NUM_OF_LOW_POWER_EVENTS=8",
> +    "QSPI_FLASH1_IO0=p1",
> +    "QSPI_FLASH1_IO1=p1",
> +    "QSPI_FLASH1_IO2=p1",
> +    "QSPI_FLASH1_IO3=p1",
> +    "QSPI_FLASH1_CSN=p1",
> +    "QSPI_FLASH1_SCK=p1"
>    ],
>    "target_overrides": {
>      "*": {
> @@ -12,7 +18,9 @@
>        "rtos.main-thread-stack-size": 32768,
>        "cordio.max-connections": 5,
>        "platform.all-stats-enabled": true,
> -      "target.mbed_app_start": "0x10000"
> +      "target.mbed_app_start": "0x10000",
> +      "target.components_add" : ["QSPIF"],
> +      "target.device_has_add" : ["QSPI"]
>      }
>    }
> ```
> 
> Of course replace `p1` with the pins you are actually using to connect the external QSPI flash

Great, I can compile the libmbed.a successfully now.
I am going to make the change and try Nano33BLE_FlashMassStorage.ino now.
Before that I have a question.
**Do I need to format the external QSPI flash before I run the Nano33BLE_FlashMassStorage.ino ???**
I ask this question because I find that the arduino edge control support the external QSPI flash and in it's relative [demo code](https://github.com/arduino-libraries/Arduino_EdgeControl/blob/master/examples/Storage/StoragePartitions/StoragePartitions.ino).
it's need to be format before use as a U disk.

facchinm, 2022-01-12:
Edge Control flash is being formatted in factory; anyway, adding the formatting phase is very easy (just add this line https://github.com/arduino-libraries/Arduino_EdgeControl/blob/master/examples/Storage/StoragePartitions/StoragePartitions.ino#L92 ) . `format()` will also mount the partition so the flow doesn't need many modifications.

closed 2022-01-12 by facchinm