arduino/ArduinoCore-mbed #378: RP2040 Core1 stops when connected via usb

MNS26, 2022-01-04:
after trying to fiddle with the pi pico trying to get core1 to "fade" the builtin led (pin 15 ) it would stop right after upload.
thinking it was a voltage issue (_unlikely but who knows_) i tried running it without usb and giving it 5v on Vusb directly, then it _did_ work.

(i already wrote something similar in a different issue but that was related to something else)

the code i used:
```
#include <mbed.h>
#include <Arduino.h>
#include "pico/multicore.h"
#include  "hardware/structs/sio.h" 
#define CORE1_LED 25u
int brightness=0;
int fadevalue = 10;
void core1_loop()
{
  while(1)
  {
  brightness=brightness+fadevalue;
  if(brightness <=0 || brightness >= 2048) {
    fadevalue = -fadevalue;
  }
    digitalWrite(CORE1_LED,HIGH);
    sleep_us(brightness);
    digitalWrite(CORE1_LED,LOW);
    sleep_us(1000);
  }
}

void setup()
{
  pinMode(CORE1_LED,OUTPUT);
  //multicore_reset_core1();
  multicore_launch_core1(core1_loop);
}

void loop()
{
}
```

MNS26, 2022-01-05:
@dlivingstone then it should also be causing issues when **not** connected via usb.

MaximumOctopus, 2022-05-01:
FWIW, I've also noticed this issue (just now). When powered via USB from a laptop, the core1 code isn't executing. When powered by 5V via VBus and GND, both cores work as expected. If I power it via the USB port from one of those cheap rechargeable battery power banks (£1 from poundstretcher) it also works as expected. 

The code I'm using is very simple. Two LEDs flashing at different rates, one per core. I'm using a framework very similar to the OP's code, plus a flasher (on, delay, off, delay) in loop().

Gerriko, 2022-05-29:
I've just had a very similar problem. I resolved by placing a long delay in setup() like so:

```
void setup()
{
  pinMode(CORE1_LED,OUTPUT);
  delay(2000);
  //multicore_reset_core1();
  multicore_launch_core1(core1_loop);
}
```

Seemed to solve my problem. I've not tested with shorter delays. I'm now curious to see if this works in other cases. 

The reason I did this was that I initially had this setup which worked fine but when I removed the while (!Serial) it stopped working. As such, I figured a delay was needed and once inserted this worked for me.

```
void setup()
{
  Serial.begin(115200);
  while (!Serial) {;;}
  multicore_launch_core1(core1_loop);
}
```

JojoS62, 2023-04-17:
I ran into the same problem, and also the workaround with the delay is working (500 ms is ok);

I've also started the debugger and the core1 stops with hardfault in strlen(). Is it a problem when the core0 is in an ISR handler when core1 starts?

[image]