arduino/ArduinoCore-mbed #379: Core 2.4.1 -> 2.5.2 may have broken MALLOC for RPI Pico?

ageppert, 2022-01-05:
I am troubleshooting some code that gives me blink errors (4 long, 5 short (I think - it's hard to count) when I compile this line in platformIO:

`Command* cmd = (Command*) malloc(sizeof(Command));`
(from https://github.com/basilfx/Arduino-CommandLine in the function CommandLine::add in the file CommandLine.cpp)

It fails with platform = raspberryPi@1.4.0 (ArduinoCore-MBED v2.5.2). 
It works with platform = raspberryPi@1.3.0 (ArduinoCore-MBED v2.4.1).

I think this is an issue at the core level, but I'm not positive.

facchinm, 2022-01-11:
Hi @ageppert ,
I just tried Arduino-CommandLine examples on core 2.6.1 and everything works as expected (compiling via Arduino IDE).
Can you provide a minimal example that crashes on 2.6.1 (either using the IDE or pio) ? Thanks

ageppert, 2022-01-22:
@facchinm Bottom line: No issue with Malloc. My mistake. I am going to close this issue. Thanks again for the support!

Explanation: I just went back and tried my complete project, which includes the "Arduino-CommandLine" code with Malloc, and it IS working on 1.3.0, 1.4.0, and 1.5.0 (and associated ArduinoCore-MBED versions). So there is not bug or problem with Malloc. I think the issue I was having was with I2C configuration and usage in my project. I've made progress on that, but still sorting it out. That progress got me away from the MBED error, and I2C config seems to be the real root of the issue I am having.

closed 2022-01-22 by ageppert