arduino/ArduinoCore-mbed #383: Action error when run the workflow

0hotpotman0, 2022-01-10:
Hi 
I download the board and run those examples all can work. but the workflow of action still reports those examples still exist problem, could you help with it, thank you.

[image]

Best regards

0hotpotman0, 2022-01-10:
this is the link : https://github.com/Seeed-Studio/ArduinoCore-mbed/runs/4749036455?check_suite_focus=true

facchinm, 2022-01-10:
Hi @0hotpotman0 ,
it looks like the issue is 
`Error during build: exec: "adafruit-nrfutil": executable file not found in $PATH` .
You may need to add a command to install that utility (via a core that provides it) in the workflow file.

closed 2022-01-10 by facchinm