arduino/ArduinoCore-mbed #385: Action release error: "cannot stat '../*.tar.bz2': No such file or directory"

0hotpotman0, 2022-01-17:
Hi   
I see your release of action always fails after version 1.3.2, 

And I check the log, there look like missed a tar.bz2 file 
[image]

Is that possible to fix?

Best regards

facchinm, 2022-01-17:
The `release` action is broken due to this line https://github.com/arduino/ArduinoCore-mbed/blob/master/extras/pack.release.bash#L28-L30 ; since the script to package the cores is really this (https://github.com/arduino/ArduinoCore-mbed/blob/master/package_full.sh ) the action is not being used at the moment and can be removed safely.

closed 2022-01-17 by facchinm

0hotpotman0, 2022-01-18:
> The `release` action is broken due to this line https://github.com/arduino/ArduinoCore-mbed/blob/master/extras/pack.release.bash#L28-L30 ; since the script to package the cores is really this [...]

I try to give a fixed version as the latest version 2.6.1, It pass that step, but the new error as below :
[image]
[image]

It looks like cannot access the server, I think it is just upload the package something to JSON and tar.bz2 file, So I try to disable it, then next step some files cannot available again

[image]

So the main problem is accessing the server to get json I think, could give me some solve, maybe some points I do not do right 

Best regards