arduino/ArduinoCore-mbed #389: Portenta H7 doesn't work with MKR Shield Dynamixel

Matt93-00, 2022-01-21:
Hi people, I found this problem during my tests. I am working on a system made from portenta h7+ mkr shield dynamixel+dynamixel. I bought the mkr shield because from the documentation https://emanual.robotis.com/docs/en/parts/interface/mkr_shield/ it is said that it is compatible with portenta, but if I run any sketch coming from DYNAMIXEL2Arduino and DYNAMIXEL Shield library, no one dynamixel is found.
Note that I know the id, baud rate, and protocol, so I can't be wrong on that. Also, the connection and the charge are good. In the end, if I use another board, for example, OpenCR, OpenCm, or Arduino Uno with the proper shield, I can drive the servo. Thus, the problem is directly the portenta I guess.
Thank you for the support.

facchinm, 2022-01-21:
Hi @Matt93-00 ,
the issue may be related with `Serial.write()` returning before the last byte has been actually sent. This breaks the RS485 communication since the last byte doesn't actually "go through" the bus. 
We tried getting this fixed on mbed level via https://github.com/ARMmbed/mbed-os/pull/14600 but it has not been merged yet.
A possible solution might be adding a delay after all the calls to `port_.flush();` in [this](https://github.com/ROBOTIS-GIT/Dynamixel2Arduino/blob/2634a23434787074869ee0705936e33dbe67fcb4/src/utility/port_handler.cpp#L110) file, but I can't test it since I don't have a shield.
The delay should be consistent with the baud rate, so if baud == 57600 the delay becomes more or less
`delayMicroseconds(180)`

Matt93-00, 2022-01-22:
Hi @facchinm,
Thank You for the interest and the tip firstly. I could test it only now, and unfortunately, it doesn't work, but I can test again other solutions if you have some other ideas. I would like to say you that I have some idea but no. 
Thank you again.

Matt93-00, 2022-01-25:
Hi again @facchinm I write you only to ask you if you can advise me as soon as you have some news. Thank You

adityakamath, 2022-04-27:
Wondering if there is any progress on this, I have the same issue. All the documentation about the MKR shield seem to indicate that it works with the Portenta H7, but it does not

ROBOTIS-Will, 2022-04-29:
I've temporarily [appended a TX delay in DYNAMIXEL2Arduino library](https://github.com/ROBOTIS-GIT/Dynamixel2Arduino/commit/2800ae9dacb6949ddacff65131ab100a1af0b26b) as the TX Enable(TX_EN or A6) become high too early before the TX packet is transmitted completely.
The delay has been tested under 9600, 57600, 115200, 1M, 2M, 3M, 4M baudrates.
Thanks!

facchinm, 2022-04-29:
Closing as patched at library level

closed 2022-04-29 by facchinm