arduino/ArduinoCore-mbed #390: wiring_analog.cpp fails to build after running ./mbed-os-to-arduino -r on Nano 33 BLE

ipener, 2022-01-21:
**OS**: Mac
**Board**: Arduino Nano 33 BLE
**IDE**: Arduino IDE Version: 2.0.0-rc3

I need to rebuild the mbed core in order to set a higher ATT MTU for BLE requests. As the [README.md](https://github.com/arduino/ArduinoCore-mbed/blob/master/README.md) suggests, I need to use `./mbed-os-to-arduino` which unfortunately doesn't work out of the box on a Mac (Bash version too low, Python3 missing, etc.). After getting It to work eventually, I'm able to compile the latest version on Mbed OS. However, I get two build failures:

```
Library/Arduino15/packages/arduino/hardware/mbed_nano/2.6.1/cores/arduino/Arduino.h:98:8: error: 'analogin_config_t' does not name a type; did you mean 'analogin_config_s'?
 extern analogin_config_t adcCurrentConfig;
        ^~~~~~~~~~~~~~~~~
        analogin_config_s
```

facchinm, 2022-02-21:
Hi @ipener ,
sorry for the late response. Some out-of-tree patches needed to adapt mbed-so to our build are contained in `patches` folder. Not all of them are strictly needed (and some have been mainlined) but the one you need is https://github.com/arduino/ArduinoCore-mbed/blob/master/patches/0004-Add-AnalogIn-configure-function.patch . Just apply it over any mbed-os version (the Nordic porting layer has not being updated in a while) and then everything should be ok.

About porting `mbed-os-to-arduino` script to run on Mac @manchoz had some patches lying around that may be useful

closed 2022-02-21 by facchinm

AllisonSimpson, 2023-03-07:
How did you manage to solve the analogin_config_t issue?