arduino/ArduinoCore-mbed #391: digitalWrite() produces weak signal on Nano 33 BLE

ipener, 2022-01-22:
**OS**: Mac
**Board**: Arduino Nano 33 BLE
**IDE**: Arduino IDE Version: 2.0.0-rc3
**Mbed OS**: Version 2.6.1

**Context**
I've correctly wired up an Arduino Nano 33 BLE to a [4-channel AC dimmer from RobotDyn](https://robotdyn.com/ac-light-dimmer-module-4-channel-3-3v-5v-logic-ac-50-60hz-220v-110v.html) and noticed that the output signal produced by calling `digitalWrite(D6, HIGH)` is way too low. The simplest code snippet to reproduce this is:

Cdo-Github, 2022-02-01:
I have a similar problem that the digitalWrite function work once only. After sending HIGH I don't get it LOW.

ipener, 2022-02-06:
Closing since this is expected behavior. Turns out that the [Arduino Nano 33 BLE](https://docs.arduino.cc/hardware/nano-33-ble) outputs 7mA (not 10mA as stated in the spec) and the [Arduino Nano 33 IoT](https://docs.arduino.cc/hardware/nano-33-iot) outputs 10mA (not 7mA as stated in the spec).

closed 2022-02-06 by ipener

facchinm, 2022-02-22:
> Closing since this is expected behavior. Turns out that the [Arduino Nano 33 BLE](https://docs.arduino.cc/hardware/nano-33-ble) outputs 7mA (not 10mA as stated in the spec) and the [Arduino Nano 33 IoT](https://docs.arduino.cc/hardware/nano-33-iot) outputs 10mA (not 7mA as stated in the spec).

We should indeed fix the documentation @sebromero