arduino/ArduinoCore-mbed #398: MKR format Portenta crashes with MKRIMU shield (possibly RESET pin issue)

hpssjellis, 2022-02-05:
I attached upwards female headers on my Arduino PortentaH7 so that the Vision shields can also plug into it using the bottom located 80 pin HD connectors. (Female headers facing up would be a good purchase option from the Arduino store) (By the way the Portenta in the following image had through hole male pins as well which when I stripped off the male pins I clipped a bit of the board and potentially have wrecked that board, my bad)

.

[image]

.

The Portenta with upwards female headers means a few of the MKR format shields that also only have upward facing female headers will not attach, but most MKR shields also have downward facing male headers so they attach fine to the Portenta. One such shield is the MKRIMU (for acceleration, roll and magnetic field)

.

[image]

.

I tried connecting these two MKR format boards together using a few of my Portenta's with and without the vision shield attached and on power up every time the PortentaH7 went into panic mode (SOLID RED LED). 

.
[image]

.

I looked at the diagram, and realized there were only a few connected pins, so using male-female breadboard wires I manually connected them only.  3.3V, GND, RESET, pin DO, pin11/SDA, pin12/SCL . Unfortunately this still caused the SOLID RED LED.

[image]

I then one at a time disconnected a wire and fired up the board, when I disconnected the RESET pin the board powered up GREEN LED. 

.
[image]

.

With the RESET pin disconnected I successfully ran the Acceleration MKRIMU example

.

[image]

.

I then tested the roll and magnetic field examples and they all worked. I then tested an advanced Machine Learning EdgeImpulse motion example  [here](https://github.com/hpssjellis/portenta-pro-community-solutions/blob/main/examples/dot5-portenta-machine-learning/dot51-portenta-edge-impulse-ml/dot512-with-accessories/dot5121-accelerometer/dot51211-WORDS/dot51211-WORDS.ino) from my library of Portenta-Pro-Community-Solutions. I  changed the EdgeImpulse standard ```#include <Arduino_LSM9DS1.h>``` for the Nano 33 Ble Sense to  ```#include <MKRIMU.h>``` for the MKRIMU and the advanced motion machine Learning program worked.

I would assume a simple fix for the MKRIMU shield to be useful on the PortentaH7 would be to snip the RESET pin, however I don't have any other MKR format boards so I don't know if this fix is detrimental to other boards. Could someone, who is not spending their own money, test this issue and possible fix and reply back. 

Understandably, I may have tweeted something in frustration about this. Happy now that it works.

hpssjellis, 2023-01-04:
Any progress on this. I even managed to orange LED brick one of my Portenta's while testing the MRK IMU

portelaraiden8, 2023-09-18:
h

portelaraiden8, 2023-09-18:
Hi hpssjellis,

i have connected that same you. So, MKRIMU does not refresh value when i move it. Do you know that what is happening?

hpssjellis, 2023-09-18:
@portelaraiden8 

Did you try bending the reset pin away before connecting it?