arduino/ArduinoCore-mbed #412: Cannot use PluggableUSBHID Keyboard and Mouse at the same time.

ApiumJacob, 2022-02-26:
The following code a USB device that does not enumerate.
```
/*
  Keyboard Mouse
*/

#include "PluggableUSBHID.h"
#include "USBKeyboard.h"
#include "USBMouse.h"

USBKeyboard Keyboard;
USBMouse Mouse;

void setup() {
  // initialize control over the keyboard:
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(1000);
  Mouse.move(100,100);
  delay(1000);
  Mouse.move(-100,-100);
  delay(1000);
  Keyboard.printf("Hello world\n\r");
  delay(5000);
}
```

Output from dmesg, complaining that descriptor is claiming 4 interfaces but only 3 are defined.
```
[183432.315758] usb 1-1.1.1: new full-speed USB device number 41 using xhci_hcd
[183432.453091] usb 1-1.1.1: config 1 has 3 interfaces, different from the descriptor's value: 4
[183432.453092] usb 1-1.1.1: config 1 has no interface number 1
[183432.453097] usb 1-1.1.1: Duplicate descriptor for config 1 interface 0 altsetting 0, skipping
[183432.454172] usb 1-1.1.1: New USB device found, idVendor=2e8a, idProduct=00c0, bcdDevice= 1.09
[183432.454174] usb 1-1.1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[183432.454175] usb 1-1.1.1: Product: RaspberryPi Pico
[183432.454176] usb 1-1.1.1: Manufacturer: Arduino
[183432.454176] usb 1-1.1.1: SerialNumber: 51D060E633998F13
[183432.469774] input: Arduino RaspberryPi Pico Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1.1/1-1.1.1/1-1.1.1:1.0/0003:2E8A:00C0.0010/input/input54
[183432.528075] input: Arduino RaspberryPi Pico Consumer Control as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1.1/1-1.1.1/1-1.1.1:1.0/0003:2E8A:00C0.0010/input/input55
[183432.528202] hid-generic 0003:2E8A:00C0.0010: input,hidraw6: USB HID v1.11 Keyboard [Arduino RaspberryPi Pico] on usb-0000:00:14.0-1.1.1/input0
[183432.528284] cdc_acm 1-1.1.1:1.2: ttyACM0: USB ACM device
```

facchinm, 2022-02-28:
Hi @ApiumJacob ,
this is a known limitation of the PluggableUSB porting for mbed-os; to declare a multi function Keyboard+Mouse HID device you should just include `USBMouseKeyboard.h`  and drop the two standalone includes

closed 2022-02-28 by facchinm

ApiumJacob, 2022-03-02:
@facchinm I'm not sure why you closed this issue, its still an issue.

facchinm, 2022-03-04:
I close it because it's not really an issue, and the OS is really telling you that something is wrong (so either the Mouse or the Keyboard will be non operational).
The correct code for mbed cores at the moment is 
```
#include "PluggableUSBHID.h"
#include "USBMouseKeyboard.h"

USBMouseKeyboard MouseKeyboard;
```
Which gives both the objects and the descriptor is correct.
Porting the PluggableHID interface to the classic one would be a very nice effort, but I'd keep it in a separate, "feature request" issue. Of course, as always, pull requests are welcome :wink:

ApiumJacob, 2022-03-06:
Fair enough and I see that this issue already exists: #252