arduino/ArduinoCore-mbed #419: New Camera class Portenta breaking changes

hpssjellis, 2022-03-09:
@sebromero 

Is there a way to make old MBED version 2.7.2 and earlier camera code work while allowing the new 2.8.0 code to also work. This would allow examples that use the camera.h library to still work for a few months until everyone has updated their boards?

pre v2.7.2 camera example. Note: 
CameraClass cam;   
uint8_t fb[320*240];  
cam.grab(fb)

```
#include "camera.h"

CameraClass cam;
uint8_t fb[320*240];

void setup() {
  Serial.begin(921600);  

  // Init the cam QVGA, 30FPS
  cam.begin(CAMERA_R320x240, 30);
}

void loop() {
  // put your main code here, to run repeatedly:
  
  // Wait until the receiver acknowledges
  // that they are ready to receive new data
  while(Serial.read() != 1){};
  
  // Grab frame and write to serial
  if (cam.grab(fb) == 0) {
     Serial.write(fb, 320*240);       
  }
  
}

```

post v2.8.0 camera example Note:    
Camera cam(himax);     
FrameBuffer fb;  
 cam.grabFrame(fb, 3000) 

```
#include "camera.h"

#ifdef ARDUINO_NICLA_VISION
  #include "gc2145.h"
  GC2145 galaxyCore;
  Camera cam(galaxyCore);
  #define IMAGE_MODE CAMERA_RGB565
#else
  #include "himax.h"
  HM01B0 himax;
  Camera cam(himax);
  #define IMAGE_MODE CAMERA_GRAYSCALE
#endif

/*
Other buffer instantiation options:
  FrameBuffer fb(0x30000000);
  FrameBuffer fb(320,240,2);
*/
FrameBuffer fb;

unsigned long lastUpdate = 0;

void blinkLED(uint32_t count = 0xFFFFFFFF)
{
  pinMode(LED_BUILTIN, OUTPUT);
  while (count--) {
    digitalWrite(LED_BUILTIN, LOW);  // turn the LED on (HIGH is the voltage level)
    delay(50);                       // wait for a second
    digitalWrite(LED_BUILTIN, HIGH); // turn the LED off by making the voltage LOW
    delay(50);                       // wait for a second
  }
}

void setup() {
  // Init the cam QVGA, 30FPS
  if (!cam.begin(CAMERA_R320x240, IMAGE_MODE, 30)) {
    blinkLED();
  }

  blinkLED(5);
}

void loop() {
  if(!Serial) {    
    Serial.begin(921600);
    while(!Serial);
  }

  // Time out after 2 seconds and send new data
  bool timeoutDetected = millis() - lastUpdate > 2000;
  
  // Wait for sync byte.
  if(!timeoutDetected && Serial.read() != 1) return;  

  lastUpdate = millis();
  
  // Grab frame and write to serial
  if (cam.grabFrame(fb, 3000) == 0) {
    Serial.write(fb.getBuffer(), cam.frameSize());
  } else {
    blinkLED(20);
  }
}

```

sebromero, 2022-03-25:
@hpssjellis Unfortunately not. But we fixed it for future versions in the new core. We added defines so you can check the core version in your code and provide different implementations for the different core versions. See here: https://github.com/arduino/ArduinoCore-mbed/blob/master/variants/PORTENTA_H7_M7/defines.txt#L85-L87
That said you could change in your examples for `CORE_MAJOR` < 3 and use the old API. It won't catch the ones who updated the core to 2.8 but if anyone was using an older core it works.

hpssjellis, 2022-03-26:
@sebromero thank you.