arduino/ArduinoCore-mbed #420: Ticker stops working after some time of use

beegee-tokyo, 2022-03-09:
Reference https://github.com/arduino/ArduinoCore-mbed/issues/235

Looks like the bug was reintroduced on a later version of the BSP.

Timers stop working again after 4294 seconds. Exactly same bug that was fixed with https://github.com/arduino/ArduinoCore-mbed/commit/a09bee8418012267fde2d8f0bbd90fa7570a23ed#diff-eedd8d3e83f8acbff34b5f185f985581038e22debedaceaf1c09139175d3050c

Can you check please?

giulcioffi, 2022-03-09:
Hi @beegee-tokyo,
I just tested the timer on a Nano RP2040 Connect with the latest core (2.8.0) and it works just fine after 4294 seconds. The [patch](https://github.com/arduino/ArduinoCore-mbed/blob/master/patches/0064-RP2040-us_ticker-fix-missing-interrupts-after-32bit-.patch) that solves this issue has not been reverted.
Which version of the core are you using?

The test sketch that I am using to validate the timer is the following:
```C++
#include "mbed.h"
#include "rtos.h"
#include <cmsis_os.h>

using namespace std::chrono_literals;
using namespace std::chrono;
using namespace rtos;
using namespace mbed;

REDIRECT_STDOUT_TO(Serial);

/** Timer for periodic sending */
mbed::Ticker test_ticker[10];
/** Semaphore to signal loop() a send request */
rtos::Semaphore sema0(1);

Thread events_thread_id;

us_timestamp_t timestamp = 1000000;

/** Thread to handle the events */
void events_thread(void);

void cb_ticker0(void)
{
  sema0.release();
}

Timeout timeout;

void setup()
{
  // Initialize Serial for debug output
  Serial.begin(115200);
  while (!Serial);

  time_t timeout = millis();
  timer_hw->timelw = 0xFFC00000;
  //timer_hw->timelw = 0xFFFFFFFF;
  timer_hw->timehw = 0;
  Serial.println(timer_hw->timelr, HEX);

  while (!Serial)
  {
    delay(250);
    if ((millis() - timeout) > 5000)
    {
      break;
    }
  }

  Serial.println("---------------------------------------------");
  Serial.println("Ticker Test");
  Serial.println("---------------------------------------------");

  Serial.println("Taking the semaphore");
  // Take all semaphores
  sema0.acquire();

  Serial.println("Starting thread to handle the events");
  events_thread_id.start(events_thread);
  events_thread_id.set_priority(osPriorityAboveNormal);

  Serial.println("Starting the tickers");
  // Start the tickers
  test_ticker[0].attach(&cb_ticker0, std::chrono::duration<time_t> {1});
  //test_ticker[0].attach_us(&cb_ticker0, timestamp);
}

void loop()
{
  Serial.print("Timerawl: ");
  Serial.println(timer_hw->timerawl, HEX);
  Serial.print("Alarm: ");
  Serial.println(timer_hw->alarm[0], HEX);
  Serial.print("Timelr: ");
  Serial.println(timer_hw->timelr, HEX);
  Serial.print("Loop awake at ");
  Serial.println(millis() / 1000);
  delay(10000);
}

extern "C" {
  extern uint32_t target_hi[4];
}

void events_thread(void)
{
  while (true)
  {
    // Wait for event
    sema0.acquire();
    Serial.print("T0 triggered at ");
    Serial.println(millis() / 1000);
    test_ticker[0].detach();
    //test_ticker[0].attach(&cb_ticker0, std::chrono::duration<time_t> {1}); 
    test_ticker[0].attach_us(&cb_ticker0, timestamp);
    Serial.println(timer_hw->alarm[0], HEX);
    //Serial.println(target_hi[0], HEX);
    Serial.println(timer_hw->timelr, HEX);
    Serial.println(timer_hw->timerawl, HEX);
    yield();
  }
}
```
Can you please double check with this sketch? Thanks

beegee-tokyo, 2022-03-09:
With which version was the patch implemented?
I am using PlatformIO and the latest version there is 1.5.0 which corresponds to v2.6.1

[image]

facchinm, 2022-03-09:
The patch was implemented in [2.3.1](https://github.com/arduino/ArduinoCore-mbed/releases/tag/2.3.1), so either platform.io build is doing something nasty (wouldn't be the first time) or your issue is slightly different.
Would you mind providing a minimal sketch that shows the problem?

beegee-tokyo, 2022-03-17:
@facchinm 

You are correct, the ticker is working. The problem is somewhere deeper in the LoRa library I am using.

closed 2022-03-17 by beegee-tokyo