arduino/ArduinoCore-mbed #422: Nicla crashed at startup with libmbed.a that built from this repo

leomonan, 2022-03-11:
Good Morning.

I tried replacing the libmbed.a with the file built from this repo, the device crashed at start up. I can't find what is wrong, can you give me a hand, thanks a lot !

I build  libmbed.a using the following commands:
```
./mbed-os-to-arduino -a NICLAL:NICLA
./mbed-os-to-arduino -r /home/ubuntu/workspace/mbed/libraries/SFU/extra/mbed-os NICLAL:NICLA
```
And replace the default arduino mbed_nicla/2.7.2/variants/NICLA/libs/libmbed.a with the new library built from above command.
Then I got crash info from both Arduino IDE's examples:
Here is the crash log:
```
++ MbedOS Fault Handler ++

FaultType: HardFault

Context:
R   0: 20000CC8
R   1: 00000000
R   2: 00000000
R   3: 0001B528
R   4: 20000CC4
R   5: 00000000
R   6: 00000000
R   7: 00000000
R   8: 00000000
R   9: 00000000
R  10: 00000000
R  11: 00000000
R  12: 00000000
SP   : 2000AE60
LR   : 00016341
PC   : 0001B528
xPSR : 00000000
PSP  : 2000AE40
MSP  : 2000FFC0
CPUID: 410FC241
HFSR : 40000000
MMFSR: 00000000
BFSR : 00000000
UFSR : 00000002
DFSR : 00000000
AFSR : 00000000
Mode : Thread
Priv : Privileged
Stack: PSP

-- MbedOS Fault Handler --

++ MbedOS Error Info ++
Error Status: 0x80FF013D Code: 317 Module: 255
Error Message: Fault exception
Location: 0x1B528
Error Value: 0x2000A90C
Current Thread: ping_thread Id: 0x20000D3C Entry: 0x16329 StackSize: 0x400 StackMem: 0x2000AA68 SP: 0x2000AE60 
For more info, visit: https://mbed.com/s/error?error=0x80FF013D&tgt=NICLA
-- MbedOS Error Info --
```

facchinm, 2022-03-11:
Hi @leomonan ,
the NICLA support package needs a fairly recent version of `mbed-os` . The one we are using to compile the core 2.8.0 is this one https://github.com/arduino/mbed-os/tree/extrapatches-6.15.1 , and you can use that branch as base via 
```
git clone  https://github.com/arduino/mbed-os -b extrapatches-6.15.1
./mbed-os-to-arduino -r path_where_you_cloned_mbed-os NICLA:NICLA
```

Let me know if it works for you! Thanks

leomonan, 2022-03-14:
Hi  @facchinm 
   I got the same startup crash error using this branch https://github.com/arduino/mbed-os/tree/extrapatches-6.15.1
   and I get error when applying patch on this branch by running command:
    `./mbed-os-to-arduino -a NICLAL:NICLA`

facchinm, 2022-03-14:
That's very strange... Which combination of core/OS/sketch are you using?

leomonan, 2022-03-14:
Yes I just run ./mbed-os-to-arduino -r path_where_you_cloned_mbed-os NICLA:NICLA 
and there is multiple defined error when linking libmbed.a to Arduino Sketch. 
So I removed this file from compile to make the link succeed.
`connectivity/nanostack/nanostack-hal-mbed-cmsis-rtos/arm_hal_random.c`

closed 2022-03-14 by facchinm

leomonan, 2022-03-15:
@facchinm That works, thank you very much!
I'm sorry for my mistake.

facchinm, 2022-03-15:
No problem @leomonan , happy to have helped :wink: