arduino/ArduinoCore-mbed #424: Example sketch `USBHID>Keyboard` makes the Arduino unusable

beyarkay, 2022-03-13:
## How the problem started
I was playing around with using the Arduino Nano BLE 33 as a keyboard, and stumbled upon the [USBHID>Keyboard sketch](https://github.com/arduino/ArduinoCore-mbed/blob/master/libraries/USBHID/examples/Keyboard/Keyboard.ino) which I got via the default suggestion from the Board Manager to install `Arduino Mbed OS Nano Boards`:

See the source reproduced below:

```c
#include "PluggableUSBHID.h"
#include "USBKeyboard.h"

USBKeyboard Keyboard;

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:
  delay(1000);
  Keyboard.printf("Hello world\n\r");
}
```

It seemed fairly innocent, so I uploaded it without a second thought thinking that I could trust anything provided from Arduino itself as an example script.

## What the problem is
But now, as soon as I plug in the Nano, it types `Hello world` followed by a newline every. single. second.

If I try to upload the blink sketch to clear it, I first need to select the Port, but the act of going through the menu to choose the correct port takes over a second, and typing in `Hello world` selects the library manager and ends up opening it when enter is pressed due to the `\n\r`. So my Arduino Nano has been rendered useless by code included as an example in the default sketches. That's not cool.

## Failed Solution Number 1
I tried creating a custom keyboard shortcut to select the port in less than a second, and then pressing `cmd-U` to upload, but I think there might also be a problem there because I get this error:

```
    ...
    Found upload port: /dev/cu.usbmodem11401
    /Users/<my username>/Library/Arduino15/packages/arduino/tools/bossac/1.9.1-arduino2/bossac -d --port=cu.usbmodem11401 -U -i -e -w /var/folders/y1/md9_k_td4rz1f0pq26d24crh0000gn/T/arduino_build_230227/<my script name>.ino.bin -R 
->  No device found on cu.usbmodem11401
    An error occurred while uploading the sketch
    ....
```

I'm absolutely certain that `cu.usbmodem11401` is the correct port, because that's what shows up when I `ls /dev/cu*` while the Arduino is plugged in. So I think that the Arduino IDE is attempting to upload to `cu.usbmodem11401`, but sees that it's a keyboard, and decides that it can't upload code to a keyboard so aborts.

## Failed Solution Number 2
I tried uploading `blink.ino` from a different laptop (also macos) but nothing changed.

## Failed Solution Number 3
I tried looking for ways to 'distrust' a keyboard to prevent the Arduino from sending keyboard strokes, but couldn't anything.

## Failed Solution Number 4
Maybe attempting to short circuit whichever wire the Arduino uses to communicate with the laptop will work, but I don't know nearly enough to do that so I'd just be randomly connecting wires to ground which I don't really want to do...

## Fin

I'm at a loss for what to try. I've posted on [arduino.stackexchange](https://arduino.stackexchange.com/questions/89039/arduino-nano-ble-33-default-example-sketch-usbhidkeyboard-makes-the-arduino-u), but I can't think of other solutions.

beyarkay, 2022-03-13:
The board was un-bricked thanks to @[majenko](https://arduino.stackexchange.com/users/4143/majenko) on the Arduino StackExchange. The solution is to double press the reset button directly after powerup, and then the Arduino will remove the current code and wait for a new sketch to be uploaded.

Leaving this issue open in hopes of the pull request getting accepted so others don't get caught out by the same problem.

facchinm, 2022-03-14:
Hi @beyarkay ,
double clicking the reset button is the default recovery method for all our boards; I hope the knowledge about this was more widespread but it looks like we are still missing to document this properly.
About the library, the original USB examples always had a "trigger" (like pressing a button or writing something on the Serial port https://github.com/arduino/arduino-examples/blob/main/examples/09.USB/KeyboardAndMouseControl/KeyboardAndMouseControl.ino#L12-L13) to enable the HID function, so I'd rework #425 to add a trigger instead.

closed 2022-03-14 by facchinm