arduino/ArduinoCore-mbed #438: [Portenta_H7] Ethernet WebServer extremely slow from core v2.7.2 - v3.0.1

khoih-prog, 2022-03-31:
#### Describe the bug

The core's Ethernet Server speed is normal for core v2.6.1-

For `core v2.7.2 to v3.0.1`, the Ethernet Server speed is **extremely slow and almost unusable**.

For example, try from complex example [**AdvancedWebServer**](https://github.com/khoih-prog/EthernetWebServer/tree/master/examples/AdvancedWebServer) to the simplest one [**WebServer**](https://github.com/arduino/ArduinoCore-mbed/tree/master/libraries/Ethernet/examples/WebServer)

#### MRE

Using the following core's Ethernet library example [**WebServer**](https://github.com/arduino/ArduinoCore-mbed/tree/master/libraries/Ethernet/examples/WebServer)  with 

- Arduino IDE v1.8.19
- mbed_portenta `core v2.6.1` or `core v2.7.2 to v3.0.1`
- Portenta_H7_M7
- Either static IP or DHCP

#### Related issue
[**[Portenta_H7] Ethernet not working in core v3.0.0** #436](https://github.com/arduino/ArduinoCore-mbed/issues/436)

facchinm, 2022-04-01:
:wave: 
I pushed https://github.com/arduino/ArduinoCore-mbed/pull/439 to fix the issue (thanks for reporting btw ! ).
To make sure we are not breaking anything it would be nice if you could test in some real world applications, both using wifi and ethernet. @hpssjellis maybe had some code lying around that could be useful for testing :wink:

khoih-prog, 2022-04-01:
The #439 fixed the slow issue for Portenta Ethernet WebServer.

But Portenta WiFi still suffers the slowness from core v2.7.2 to v3.0.1, even with the #439 fix.

Core v2.6.1 is also tested and OK with Portenta WiFi.

### To test

1. Install [**WiFiWebServer releases v1.4.0**](https://github.com/khoih-prog/WiFiWebServer/releases/tag/v1.4.0)
2. Using AdvancedWebServer example then test with core v2.6.1, then core v2.7.2+

I'll open a new issue for this Portenta WiFi.

closed 2022-04-01 by khoih-prog

hpssjellis, 2022-04-01:
@facchinm 

> To make sure we are not breaking anything it would be nice if you could test in some real world applications, both using wifi and ethernet. @hpssjellis maybe had some code lying around that could be useful for testing 😉

Using Arduino MBED 3.0.1 a student confirmed both the WiFi and Ethernet minimal examples are working as you mentioned. I am presently testing adafruit.io  MQTT and HTTP (POST, GET) Webhook clients, so I should have more information next week. 

I will probably be using the advanced async clients made by @khoih-prog, but open to suggestions for simple blocking  MQTT and HTTP (POST GET) clients as well.

PR #439 (2022-04-28): Drastically speedup client.read() operation
Since the data being consumed by the reader thread is going to be read somewhere via read() APIs (that are known to be fair and call yield()), we can avoid the delay() when the buffer needs to be flushed by setting the reader thread priority to something lower than the main thread.

To be tested on some real world applications (by now, tested with Ethernet/WiFi Client/Server "basic" examples)

Fixes #438 

@manchoz