arduino/ArduinoCore-mbed #441: [Portenta_H7] WiFi WebServer extremely slow from core v2.7.2 - v3.0.1

khoih-prog, 2022-04-01:
The #439 fixed the slow issue for Portenta Ethernet WebServer.

But Portenta WiFi still suffers the slowness from core v2.7.2 to v3.0.1, even with the #439 fix.

Core v2.6.1 is also tested and OK with Portenta WiFi.

### To test

1. Install [**WiFiWebServer releases v1.4.0**](https://github.com/khoih-prog/WiFiWebServer/releases/tag/v1.4.0)
2. Using AdvancedWebServer example then test with core v2.6.1, then core v2.7.2+

khoih-prog, 2022-04-01:
@hpssjellis

https://github.com/arduino/ArduinoCore-mbed/issues/438#issuecomment-1085557092

> To make sure we are not breaking anything it would be nice if you could test in some real world applications, both using wifi and ethernet. @hpssjellis maybe had some code lying around that could be useful for testing :wink: 

Please help testing using real-world applications, if possible. I'm currently tied up with some other things. Thanks.

facchinm, 2022-04-05:
Hi @khoih-prog ,
I tested WiFiServer with the latest patch and it looks like the issue is solved, as with Ethernet.
Are you sure you are testing the correct branch for WiFi too?

khoih-prog, 2022-04-05:
Hi @facchinm 

I retested and found out this is an issue of my [WiFiWebServer library](https://github.com/khoih-prog/WiFiWebServer), noy keeping up with the recent changes in the Portenta_H7 core.

Already found out and will fix the library in next release.

[image]

Thanks,

closed 2022-04-05 by khoih-prog

hpssjellis, 2022-04-06:
Question: Portenta WiFi has WiFiSSLClient, but Portenta Ethernet does not seem to have EthernetSSLClient. Is that correct and is it in process or am I missing something.

On a positive side: I have WiFI connections with Adafruit for both secure MQTT and HTTP working. I also have Ethernet insecure Adafruit  HTTP working and am working on Ethernet MQTT.