arduino/ArduinoCore-mbed #442: PDM sound with high sizzle on RP2040

nerodu, 2022-04-02:
Hi, 
I'm working a project with a stereo PDM microphone(MP34DT06J x 2) with RP2040. Thanks for the PDM library, and I get record voice now. However, I find the voice has high frequence sizzle noise when play. When record, I am in quiet environment. So I think maybe I can modify some parameters of OpenPDMFilter.c. But it looks difficult for me, because I am not clear with the meaning of each parameters... Could you give some suggestion about this~~

nerodu, 2022-04-02:
[image]

This is a quiet record without voice. left and right channel looks have many noise. Does the library support wave filter?

cdowd9556, 2023-03-31:
were you using different pio program to multiplex stereo pdm data on one pin?