arduino/ArduinoCore-mbed #446: UpdateBootloader never outputs on terminal - can't update bootloader through sketch

bashynx, 2022-04-06:
I am using portenta H7 with vision shield - LoRa

Trying to run this two codes didn't work for me:
- https://github.com/arduino/ArduinoCore-mbed/blob/master/libraries/STM32H747_System/examples/STM32H747_updateBootloader/STM32H747_updateBootloader.ino
- https://github.com/arduino/ArduinoCore-mbed/blob/master/libraries/STM32H747_System/examples/STM32H747_getBootloaderInfo/STM32H747_getBootloaderInfo.ino

After successfully uploading the file, I never saw the output on terminal. 

When uploading different sketches, they work, just this one doesn't (I even tried to leave terminal opened for around 1 minute, just in case it will take long time to pop up)

Running the following code through different sketch:
```
  uint8_t* bootloader_data = (uint8_t*)(0x8000000 + 0x1F000);
  uint8_t currentBootloaderVersion = bootloader_data[1];
  Serial.println("Bootloader version: " + String(currentBootloaderVersion));
```
Resulted in me learning I am running bootloader v 21.

Now, I am unsure if this is my problem, or it's an issue with the new core update.

pennam, 2022-04-08:
@bashynx is the board crashing (red LED blinking) or it just not outputs anything on serial? I've made a test with the same hardware setup and bootloader version and it is working fine here so maybe i am missing something.

Could you please retry and if it is still not working, make a test commenting out this lines from the sketch https://github.com/arduino/ArduinoCore-mbed/blob/9eeba8d63e957fec1eba81b29299d1af6d856ad4/libraries/STM32H747_System/examples/STM32H747_getBootloaderInfo/STM32H747_getBootloaderInfo.ino#L2 https://github.com/arduino/ArduinoCore-mbed/blob/9eeba8d63e957fec1eba81b29299d1af6d856ad4/libraries/STM32H747_System/examples/STM32H747_getBootloaderInfo/STM32H747_getBootloaderInfo.ino#L8-L13 

Could you also copy paste the full sketch you are using to get your bootloader version?

bashynx, 2022-04-08:
By trial and error, I've found out, that I need to put some `println` into the loop.

All the messages show when first message in loop gets send.

[image]

Now, I used this sketch:
``` C
uint8_t* bootloader_data = (uint8_t*)(0x801F000);
uint8_t* bootloader_identification = (uint8_t*)(0x80002F0);

void setup() {  
  Serial.begin(115200);
  while (!Serial) {}

  uint8_t currentBootloaderVersion = bootloader_data[1];
  String currentBootloaderIdentifier = String(bootloader_identification, 15);

  if(!currentBootloaderIdentifier.equals("MCUboot Arduino")) {
    currentBootloaderIdentifier = "Arduino loader";
  }

  Serial.println(currentBootloaderIdentifier);
  Serial.println("Magic Number (validation): " + String(bootloader_data[0], HEX));
  Serial.println("Bootloader version: " + String(bootloader_data[1]));
  Serial.println("Clock source: " + getClockSource(bootloader_data[2]));
  Serial.println("USB Speed: " + getUSBSpeed(bootloader_data[3]));
  Serial.println("Has Ethernet: " + String(bootloader_data[4] == 1 ? "Yes" : "No"));
  Serial.println("Has WiFi module: " + String(bootloader_data[5] == 1 ? "Yes" : "No"));
  Serial.println("RAM size: " + String(bootloader_data[6]) + " MB");
  Serial.println("QSPI size: " + String(bootloader_data[7]) + " MB");
  Serial.println("Has Video output: " + String(bootloader_data[8] == 1 ? "Yes" : "No"));
  Serial.println("Has Crypto chip: " + String(bootloader_data[9] == 1 ? "Yes" : "No"));
}

String getUSBSpeed(uint8_t flag) {
  switch (flag){
  case 1:
    return "USB 2.0/Hi-Speed (480 Mbps)";
  case 2:
    return "USB 1.1/Full-Speed (12 Mbps)";
  default:
    return "N/A";
  }
}

String getClockSource(uint8_t flag) {
  switch (flag){
  case 0x8:
    return "External oscillator";
  case 0x4:
    return "External crystal";
  case 0x2: 
    return "Internal clock"; 
  default:
    return "N/A";
  }
}

void loop() {
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(2500);                       // wait for a second
  Serial.println("ON");
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  delay(2500);                       // wait for a second
  Serial.println("OFF");
}
```

pennam, 2022-04-11:
@bashynx are you using Windows or Linux?

if in the original sketch you substitute 
```
while (!Serial) {}
```
with
```
  int start = millis(); 
  while (!Serial && millis() - start < 5000) {}
  ```
 and wait 5 seconds, does the output appears on the serial monitor?

bashynx, 2022-04-11:
I'm using windows.

But considering I made this work in the loop, I think we can close the issue? (If I put println in the loop(), I get the whole info wall)