arduino/ArduinoCore-mbed #450: Error Compiling: 1.8.0-arduino/imgtool": file does not exist

AntaresAdroit, 2022-04-08:
Hello.

I recently installed the "Arduino Mbed OS RP2040 Boards" package from the Board Manager to use with a Raspberry Pi Pico board. When I try to compile the code, I get the following error:
`exec: "C:\\Users\\Kris Kringle\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\imgtool\\1.8.0-arduino/imgtool": file does not exist
Error compiling for board Raspberry Pi Pico.`

I have tried this on two different computers, one running version 1.8.16 of the IDE and one running 1.8.19. 

The path `C:\Users\Kris Kringle\AppData\Local\Arduino15\packages\arduino\tools\imgtool\1.8.0-arduino` exists, but there is only a single file (`LICENSE.txt`) in there.

When I look for "imgtool" in this repository, there are mentions in 3 different files:

- [extras/package_index.json.NewTag.template](https://github.com/arduino/ArduinoCore-mbed/blob/79b31d9af38353112a0dee05dcef34fd0f58e1d9/extras/package_index.json.NewTag.template)
- [/boards.txt](https://github.com/arduino/ArduinoCore-mbed/blob/9ae4d85227c0417b204cdf8e0a318b0d73611fa9/boards.txt)
- [/platform.txt](https://github.com/arduino/ArduinoCore-mbed/blob/e0b58b4e3c439e7ac4866aa426fd97ecb76857b9/platform.txt)

(From this search: https://github.com/arduino/ArduinoCore-mbed/search?q=imgtool )

Am I missing a dependency?

AntaresAdroit, 2022-04-08:
Changing to version 2.8.0 of the "Arduino Mbed OS RP2040 Boards" package allows sketches (I'm using the Blink sketch to test with) to compile and execute on the Pico board as expected.

The path (`C:\Users\Kris Kringle\AppData\Local\Arduino15\packages\arduino\tools\imgtool\1.8.0-arduino`) appears unmodified. It exists, but there is only a single file (LICENSE.txt) in there.

I do see some recent commits that mention imgtool like this one: 

https://github.com/arduino/ArduinoCore-mbed/commit/18a09b35343f2403f59b614e2384bd6d92f42e97

pennam, 2022-04-11:
Hi @AntaresAdroit,
it looks like you are missing the `imgtool` binary that is coming together with the `LICENCE.txt` file from this [package](http://downloads.arduino.cc/tools/imgtool_1.8.0-rc1_Windows_32bit.zip) or [this](http://downloads.arduino.cc/tools/imgtool_1.8.0-rc1_Windows_64bit.zip) depending on your system.

Since the `LICENCE.txt` file is correctly in place, but the binary is missing, i'm wondering if it is possible that the file has been removed by your antivirus.

I've made a test on my windows VM using a fresh install of the IDE 1.8.19 and it looks like imgtool is correctly installed as dependency of the  "Arduino Mbed OS RP2040 Boards" package (version 3.0.1)

AntaresAdroit, 2022-04-11:
Thanks for your help. The code does indeed compile successfully if copy the executable binary from the zip archive you linked to (the 64 bit version in my case) with the 3.0.1 version of the board package. And indeed a fresh install of the IDE with the 3.0.1 version of the package also works for me (on a 3rd computer).

I didn't notice any antivirus messages extracting the executable manually, but both of the computers I noticed the issue with are running the same antivirus software. I also have a lot of the same boards packages installed on those two computers, so I wonder if I have some other package that incorrectly installed the `imgtool` binary in a way that caused some dependency check to fail. I think I'll do a few more experiments to see if I can find something reproducible. 

I'm curious what that tool actually is. Is it this mcuboot project? https://github.com/mcu-tools/mcuboot

pennam, 2022-04-12:
@AntaresAdroit  yes, the tool is coming from the mcuboot project. `imgtool` is actively used only by the "Arduino Mbed OS Portenta Boards" and it can be used to generate signed and encrypted binary updates. For all the other mbed platforms the builder invokes `imgtool exit` and it doesn't do anything on the binary, but this is needed because the post-build step is common within all the mbed platforms.

You can find more about the project here https://github.com/arduino/mcuboot-arduino-stm32h7

AntaresAdroit, 2022-04-12:
I checked the log for the Sophos Antivirus software on my computer and, sure enough, there is a hit there for `imgtool.exe` as a "Generic Potentially Unwanted Application". 
And I realized while watching the folder structure during the install of other board packages (trying to recreate the issue on another computer) that my suggestion that some other package that incorrectly installed the `imgtool` binary is not really possible. 
Thanks for your help.

closed 2022-04-12 by AntaresAdroit

per1234, 2022-04-13:
Thanks so much for your report @AntaresAdroit.

I have confirmed that some security tools are detecting the Windows build of [the "1.8.0-arduino" version](https://github.com/arduino/imgtool-packing/releases/tag/1.8.0-arduino) of the imgtool binary we distribute with the 3.0.0 and 3.0.1 releases of the "**Arduino Mbed OS \_\_\_ Boards**" platforms as malware:

https://www.virustotal.com/gui/file/ee969757ac819d2582b1dd36ba7ea2748d39968775b01bead2eba47f92b73fb9

I will notify Arduino's security team about this.

CC: @arduino/security @rhpco @rhaidiz @umbynos