arduino/ArduinoCore-mbed #453: Is deep sleep supported on Edge Control ?

brinata, 2022-04-12:
Hi,

I've tried to enable the deep sleep condition on the Edge Control Board, but the results are bad and not different from the normal sleep function: the board consumption doesn’t fall under 3 mA, very far from the value of 200uA declared in the technical sheet.

Is deep sleep implemented ?
Is the any example ?

Regards

manchoz, 2022-05-30:
Hi @brinata,
Please, take a look at [LowPowerButton.ino](https://github.com/arduino-libraries/Arduino_EdgeControl/blob/master/examples/Basic/LowPowerButton/LowPowerButton.ino).

brinata, 2022-05-30:
I've already tried that example, but the measured current consumption is still high ( ~ 500uA) . Moreover, if I switch on and off the i2c expander, the current consumption doesn't drop under 800 uA: Why ???? 

That's a tragedy in my application because the CR2032 always supply the CPU and so after a while date and time are lost !!! ( we have an application with a seasonal usage....)

When will the hardware RTC be available ????

Regards

manchoz, 2022-05-31:
Please,
be sure that every device connected to the board is also powered-off and that all the relevant GPIOs are in their respective low power mode.

Regarding the I/O Expander, the [::end()](https://github.com/arduino-libraries/Arduino_EdgeControl/blob/5ecab3907d4f6aa0252a9d6866dccd3f939f6c62/src/EdgeControl_Expander.cpp#L41) method will put the board in the status the expander was before its [::begin()](https://github.com/arduino-libraries/Arduino_EdgeControl/blob/5ecab3907d4f6aa0252a9d6866dccd3f939f6c62/src/EdgeControl_Expander.cpp#L14) to not disrupt any previous global powering status: please, check the order you are calling the end() methods.

Also, the board is designed to support low power mode in the case of powering from a 12V lead-acid battery, not from the CR2032 one, which is meant only for retaining RTC (both as nRF5240 internal Real-Time *Counter* for old version or proper Real-Time Clock for the new one).

manchoz, 2022-05-31:
Hi,

> At the beginning, I've just testes the example " as is " and the current that I measure was 450 uA from the 12v battery terminal. What did you measured during you test ?

This is the expected power consumption in low power mode.

Please, remember to call the `delay()` deliberately/as often as possible, because it will also put the board in deep sleep mode (mbedOS, which the ArduinoCore-mbed is based on, works in low-power mode by design).

> 
> About the expander I will check.
> 
> About the RTC, using the internal Real Time Counter is quite uncomfortable because when the board should be switched off, actually is running so I have to use the POWER_ON input to put the board in deep sleep to avoid the CR2032 battery fault in just few days. Moreover, the RTC battery should be last years, not month or weeks: as I wrote before, we have a seasonal agriculture application and I can't replace the battery every time before switch on the equipment ...

The new version of the board is able to stay powered for a few months on the CRC2032 when using a "ship mode" [sketch](https://github.com/arduino-libraries/Arduino_EdgeControl/blob/master/examples/Basic/RealTimeClock_LowPower/RealTimeClock_LowPower.ino) (set/print RTC clock and go to a `delay()`-only `loop()` function). Anyway, the CR2032 battery is not meant to be the main power supply of the board: the board is designed to be powered via a 18V solar panel or a 12V lead-acid battery (or both).

> 
> a question: I need to make more question about the board: can I make them here ? do I have to open a new post ?

Please, post them on the [Edge Control](https://github.com/arduino-libraries/Arduino_EdgeControl/issues) repository.

> 
> Regards

brinata, 2022-05-31:
at the moment using all the nominal capacity of the CR2032 ( 220 mAh) with a current consumption of 0.45mA, the battery lasts non more 20 day... not so much.

I perfectly know that a CR2032 battery has to be mainly used only for backup, but  looking the schematic present on the Arduino store it's clear that the battery supply the MCU when the 18V solar panel or a 12V lead-acid battery are missing. Normally the battery supply only the RTC circuit and/or a small backup memory, not more and normally the backup function lasts years ( the PCF8563 need about 0.25 uA, so with a CR2032 battery it can retain data for a long time ...).

At the moment I developing with the old version of the board, so when will the new version be available for selling ??? for us in very important because if the time to market will be long most probably we will consider to change hardware ..

Goliath86, 2024-01-26:
Hi, I'm using the Edge Control board recently purchased. I'm also trying to get it to work in deep sleep and I too noticed that once the Expander was turned on and then off (with the ::end() function) the board consume ~ 800uA (the loop function is a long delay as suggested). Things get worse if you connect an MKR board: I've connected a MKR GSM 1400 and with this board connected on the MKR1 slot and the power turned off (Power.off(PWR_MKR1)) the current consumption is ~ 2mA; if I disconnect the MKR board even when the firmware is running, the current consumption drop to ~ 800uA (if the Expander is turned on and then off). After checking the schematics I think that the pull-up resistors of the MKR board on the I2C bus are what cause that high current consumption because at the VIN of the MKR1 slot, even if the slot power is turned off, I read ~ 1.5V (I don't know why)