arduino/ArduinoCore-mbed #454: digitalRead() analog pin A1

msHTWK, 2022-04-13:
I try to read out the analog pins A0 to A5 like its supported from arduino. But when i try to read A1 as digitalRead() nothing happend.
Here is quick example and during the routine  i hook up 5V on A1...

// the setup routine runs once when you press reset:
void setup() {
  // initialize serial communication at 9600 bits per second:
  Serial.begin(9600);
}

// the loop routine runs over and over again forever:
void loop() {
  // read the input on analog pin 0:
  int sensorValue = analogRead(A1);
  int sensVal = digitalRead(A1);
  // Convert the analog reading (which goes from 0 - 1023) to a voltage (0 - 5V):
  float voltage = sensorValue * (5.0 / 1023.0);
  // print out the value you read:
  Serial.println(voltage);
  Serial.println(sensVal);
  delay(500);
}

Its working for A0,A2,A3,A4 and A5 but not for A1
maybe someone has an Idea or a work around

facchinm, 2022-04-13:
Hi @msHTWK ,
hooking up 5V to any pin on the Portenta might have destructive results (most pins are rated up to 3.3V ! ) . Moreover, `digitalRead()` on A0 to A3 is not supported completely since that pins are "shared" with other digital pins via an internal analog switch, so the readings are almost always wrong. @sebromero do you have an handy link with this kind of documentation?

closed 2022-04-13 by facchinm

msHTWK, 2022-04-19:
Thanks for that answer. But is there any upgrade planned that these pins are useable in future?