arduino/ArduinoCore-mbed #461: Arduino RP2040 RTC is undefined

markub3327, 2022-04-17:
Hi,

   linker cannot access the hardware/rtc.h methods, `rtc_get_datetime(datetime_t*)` and `datetime_to_str(char*, unsigned int, datetime_t const*)`. 

**RTC_Demo.ino**
```cpp
#include "hardware/rtc.h"
#include "pico/util/datetime.h"

char datetime_buf[256];
char *datetime_str = &datetime_buf[0];

// Start on Friday 5th of June 2020 15:45:00
datetime_t t = {
            .year  = 2020,
            .month = 06,
            .day   = 05,
            .dotw  = 5, // 0 is Sunday, so 5 is Friday
            .hour  = 15,
            .min   = 45,
            .sec   = 00
};

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
  while(!Serial);

  // Start the RTC
  _rtc_init();
  rtc_set_datetime(&t);
}

void loop() {
  // put your main code here, to run repeatedly:
  rtc_get_datetime(&t);
  datetime_to_str(datetime_str, sizeof(datetime_buf), &t);
  Serial.println(datetime_str);
  sleep_ms(100);
}
```

## Result
```shell
/private/var/folders/_t/2r3zl5f929b2xz8ts0kfhfgr0000gn/T/arduino-sketch-DF3E31D1448C4ECE982D7DDD3165ED00/sketch/RTC_Demo.ino.cpp.o: In function `loop':
/Users/martin/Documents/Arduino/RTC_Demo/RTC_Demo.ino:31: undefined reference to `rtc_get_datetime(datetime_t*)'
/Users/martin/Documents/Arduino/RTC_Demo/RTC_Demo.ino:32: undefined reference to `datetime_to_str(char*, unsigned int, datetime_t const*)'
/private/var/folders/_t/2r3zl5f929b2xz8ts0kfhfgr0000gn/T/arduino-sketch-DF3E31D1448C4ECE982D7DDD3165ED00/sketch/RTC_Demo.ino.cpp.o: In function `setup':
/Users/martin/Documents/Arduino/RTC_Demo/RTC_Demo.ino:25: undefined reference to `_rtc_init()'
/Users/martin/Documents/Arduino/RTC_Demo/RTC_Demo.ino:26: undefined reference to `rtc_set_datetime(datetime_t*)'
collect2: error: ld returned 1 exit status
```

josefwegner, 2022-04-17:
Hi,
This is a known issue. The platform is still using a rather old version of the pico-sdk, and some definitions are missing an `extern "C"` guard.

Please see https://github.com/arduino/ArduinoCore-mbed/issues/300

As a workaround, please add the following to one of your header or source files:
```
#ifdef __cplusplus
extern "C"
{
  void rtc_init(void);
  bool rtc_get_datetime(datetime_t *t);
  bool rtc_set_datetime(datetime_t *t);
  void datetime_to_str(char *buf, uint buf_size, const datetime_t *t);
}
#endif
```

markub3327, 2022-04-19:
Also, please can you inform about this issue in your official documentation. Thanks.

khoih-prog, 2022-04-19:
You can have the better workaround as in 

[**RP2040_RTC**.h#L42-L56](https://github.com/khoih-prog/RP2040_RTC/blob/main/src/RP2040_RTC.h#L42-L56)

```
extern "C" 
{
  typedef void(* rtc_callback_t )(void);
  
  void rtc_init           (void);
  bool rtc_set_datetime   (datetime_t *t);
  bool rtc_get_datetime   (datetime_t *t);
  void rtc_set_alarm      (datetime_t *t, rtc_callback_t user_callback);
  void rtc_disable_alarm  (void);
  void rtc_enable_alarm   (void);

  bool rtc_alarm_repeats(datetime_t *t);

  void datetime_to_str(char *buf, uint buf_size, const datetime_t *t);
}
```

markub3327, 2022-04-29:
It's working for me.

Thanks.

Workshopshed, 2022-06-25:
A couple of examples of using the Pico RTC can be found here - https://github.com/Workshopshed/PiPico