arduino/ArduinoCore-mbed #466: Why does an Arduino Nano 33 BLE Blinky HEX file require SoftDevice

Gerriko, 2022-04-23:
I probably shouldn't be trying this but I did anyway... ;-)

I have an nRF52840 dongle, which requires either the nRFConnect app or the command line interface to flash firmware.

I wanted to see what happens if I took a HEX file compiled by Arduino IDE using the Arduino Nano 33 BLE board configuration and then used the nRFConnect app to flash this firmware onto the nRF52840 dongle.

So I compiled the Arduino blinky example, as "LED_BUILTIN" is GPIO P0_13, which is broken out on the nRF52840 dongle.

I also knew that the Nordic nRF52 SDK's blinky example works fine when flashed via the nRFConnect app. All you do is select the hex file and the app does the rest to compile.

So I was surprised to see that when you select the Arduino Blinky HEX file, the nRFConnect app then asks for the appropriate softdevice to be included... as it somehow detects this within the HEX file. 

This shouldn't be the case as within defines.txt there is a setting DTARGET_SOFTDEVICE_NONE.

So I was curious to find out if this is purely a compatibility issue, or if there is something that is not right, I thought it best to flag this as issue.

hpssjellis, 2022-04-24:
@Gerriko I don't really think what your trying to do is either an Arduino or Nordic issue as both groups really only support their own boards.

I messed around with the NRF52840 dongle and the nano33BleSense with some success [here](https://community.platformio.org/t/like-if-you-have-an-nrf52840-nordic-usb-dongle-arduino-nano-33-ble-adafruit-sparkfun-april-makerdairy-fanstel-particle-or-any-other/11112/8)   but have since moved on to better boards such as the Portenta as what I was doing proved to be quite a bit of work and then you lost access to a few pins. The particle debugger really helped my loading of the hex files.

Gerriko, 2022-04-24:
@hpssjellis thanks for the reply. The reason for flagging was to make sure. 

Yes I came across your link (didn't fully understand it, mind you). The problem is, I prefer using offline IDE's to PlatformIO and your docker solution was not suitable to my needs. Also I was not looking to change the dongle bootloader, and if I did I would prefer to use the Nano 33 BLE version to Adafruit's nRF52 bootloader option.

Based on the Nordic documentation for the Dongle (https://infocenter.nordicsemi.com/topic/ug_nrf52840_dongle/UG/nrf52840_Dongle/intro.html) their bootloader is open and you can also use their nRF Util application to flash your firmware. So I don't see the need to change this bootloader. Instead, I am looking to find a way to incorporate it within Arduino and use a modified version of the Arduino Nano 33 BLE package to program as it is almost a clone of the dongle. In fact, according to documentation, if you add the Nordic DFU Trigger Library (USB) to your custom device firmware, you can trigger the DFU without using any buttons. 

So whilst the Portenta board is an excellent board, it is not needed for my simple task. 

All I am trying to do, at this stage, is blink an LED and then eventually get the dongle to read a UART TX stream using Arduino as the IDE. I can readily do this using Segger Embedded Studio, or I could readily use my Arduino Nano 33 BLE board too. However, I wanted to explore the option of using Arduino IDE for the Dongle as it's pretty easy to modify the Nano 33 BLE board mbed config files etc.

You've got start somewhere...

closed 2022-04-24 by Gerriko