arduino/ArduinoCore-mbed #467: wrapper for MBED arduinos to use code written for Arduino SD library

hpssjellis, 2022-04-25:
Has anyone figured out a wrapper for Arduino MBED  SD card examples (I use the Portenta) to work with traditional Arduino SD.h library [here](https://github.com/arduino-libraries/SD)    

Specifically my issue is I am trying to save images from the Portenta camera to SD card using the PNGenc library [here](https://github.com/bitbank2/PNGenc)   using the following reference sketches but not having much luck.

@sebromero 

OR has anyone got example code that stores the Portenta Vision Shield Grayscale 320x320 Camera data to the Portenta Vision Shield SD card? (I guess this would be the best starting point)

Reference Files

my Portenta SD Card saving utilities  [dot513-utilities](https://github.com/hpssjellis/portenta-pro-community-solutions/tree/main/examples/dot5-portenta-machine-learning/dot51-portenta-edge-impulse-ml/dot513-utilities)

my Portenta Camera OLED files [here](https://github.com/hpssjellis/portenta-pro-community-solutions/tree/main/examples/dot3-portenta-vision-shields/dot35-camera-and-oled)

Official Arduino Portenta SD Card example [here](https://github.com/arduino/ArduinoCore-mbed/blob/master/libraries/Portenta_SDCARD/examples/TestSDCARD/TestSDCARD.ino)

Official Arduino Raw Camera example [here](https://github.com/arduino/ArduinoCore-mbed/blob/master/libraries/Camera/examples/CameraCaptureRawBytes/CameraCaptureRawBytes.ino)

closed 2022-05-24 by hpssjellis

hpssjellis, 2022-05-24:
Hi @sebromero I actually have had more problems, tried saving camera results as PNG files and could not do it so I updated the feature request here 

[https://github.com/arduino/ArduinoCore-mbed/issues/482](https://github.com/arduino/ArduinoCore-mbed/issues/482)

sebromero, 2022-05-25:
@hpssjellis Alright. It makes sense to document that. I'll put it in our backlog. Bear with us in the meantime :-)