arduino/ArduinoCore-mbed #469: Error compiling:

rmlearney, 2022-04-25:
Perhaps linked to #450 

VSCode 1.66.2 on MacOS 12.3.1, M1 MacBook Pro

Successful compilation for nano33 calls `{runtime.tools.imgtool.path}/imgtool exit` at the end. Reading #450 this seems expected.

Then after a load of 'Using library ....' messages (also expected) it returns:

`Error during build: fork/exec {runtime.tools.imgtool.path}/imgtool: no such file or directory
IntelliSense auto-configuration disabled.
[Error] Verifying sketch 'RenoSure_Final.ino': Exit with code=1`

and borks

pennam, 2022-04-26:
Hi @rmlearney
i didn't understood exactly what changed between the successful build and the broken one, but answering your question, yes and no.

Yes: starting from release 3.0.0 of the mbed core imgtool is requested, for a succesful build, from all the mbed boards not only for Portenta

No: You don't need to install it manually becuase it is automatically installed by the board manager when you install or upgrade the core.

Is imgtool present in the tools directory of your Arduino15 folder?
Which board/mbed core are you trying to use?

rmlearney, 2022-04-26:
Hi @pennam I originally cloned https://github.com/arduino/ArduinoCore-mbed.git and keep my local updated as my 'board file'. 

Going back into the Arduino GUI the board manager recognises I'm at 3.0.1, but I don't have imgtool installed anywhere.

Can I pull imgtool from somewhere else?

pennam, 2022-04-26:
You can download it manually from here https://github.com/arduino/imgtool-packing/releases/tag/1.8.0-arduino and place it in 

`Arduino15/packages/arduino/tools/imgtool/1.8.0-arduino/imgtool`

You should get the same result installing  or updating another mbed core from the GUI, for example the "Mbed OS Portenta boarsds" or "Mbed OS RP2040 boards".

In which folder did you clone the [core](https://github.com/arduino/ArduinoCore-mbed.git) ?

rmlearney, 2022-04-26:
That repo seems to be private :(

I cloned it into `/Users/rmlearney/Documents/Arduino/`

umbynos, 2022-04-26:
Now should be public

pennam, 2022-04-26:
Thanks @umbynos

rmlearney, 2022-04-26:
One final note @umbynos @pennam you might want to add these instructions to the `README.md` in https://github.com/arduino/ArduinoCore-mbed

per1234, 2022-04-26:
Even though it is valuable to have access to the imgtool alone from the release assets at the `arduino/imgtool-packing` repository, I think the best approach in general for easily getting the full toolchain in the correct places is to install the latest release version of the platform (even if you aren't planning to use that version of the platform itself).

This is the approach used in the examples compilation "smoke test" CI workflows in all the official Arduino boards platform repositories ([example](https://github.com/arduino/ArduinoCore-mbed/blob/0c042075196b67d5ef2f20d824e3a49800fe9b10/.github/workflows/compile-examples.yml#L139-L140)) and I use it myself for beta testing.

That approach is mentioned in the readme already:

https://github.com/arduino/ArduinoCore-mbed#test-things-out

>To install ARM build tools, use the Boards Manager option in the Arduino IDE to add the Arduino mbed-enabled Boards package.

However, I think there is still probably room for improvement in the documentation. I would add that as one of the steps in the installation rather than as a troubleshooting item after the user has already encountered a confusing error.