arduino/ArduinoCore-mbed #474: Bizarre problem with the Portenta H7

GunnerDx88, 2022-05-05:
Greetings,

Yesterday I was trying to debug on the portenta H7 using the ST-Link. For that, I used the JTAG conenctors + I had the usb cable always connected to the board (not sure if this is okay or not but otherwise the ST-Link was failing). I was able to reprogram the card via the ST-Link yesterday and everything was working fine. This morning I turned on my laptop and I noticed that the RGB led was not working instead the Battery Charger Led on the board was on and on top of that my PC no longer recognizes the Portenta H7 card. It's like it's not even connected. Does anyone know what may be the problem here ? Or are there any possible solutions ?
Any help is appreciated.

Thank you.

pennam, 2022-05-05:
You have to try to connect while keeping the board under reset and then releasing. You can use [ST-LINK utility](https://www.st.com/en/development-tools/stsw-link004.html) if you are on Windows enabling the connect-under-reset option in the settings menu.

pennam, 2022-05-05:
to connect while keeping the board under reset: press the reset button, connect the usb cable keeping the reset button pressed and connect using ST-LINK utility (or openocd) and then release the reset button.

GunnerDx88, 2022-05-06:
That did the trick. Thank you so much.

pennam, 2022-05-06:
Great!

closed 2022-05-06 by pennam