arduino/ArduinoCore-mbed #476: Debugging the portenta using the ST-Link

GunnerDx88, 2022-05-09:
Greetings,

I'm trying to debug the Portenta H7 board using the ST-Link probe and Sloeber as IDE.
I'm not even sure if that is possible or not. I figured that it would be possible since I'm able to debug projects on the board using the STM32Cube IDE and since Sloeber and CubeIDE are both eclipse based, I'm thinking there is no reason why it shouldn't work in Sloeber as well.
The problem is that, in CubeIDE I don't have to configure the debugger but in Sloeber I have to manually do it.
Does anyone know if this is possible or not ? If so, can you please help me with the steps to do it ?

Thank you.

GunnerDx88, 2022-05-10:
Can anyone please tell me if this is even possible or not ? That's all I want to know

facchinm, 2022-05-11:
Hi @GunnerDx88 ,
I have to experience in Sloeber but if it provides a way to specify the elf file to be debugged and an openocd commandline it will work for sure!
For reference, the commandline we use with an stlink is
```
~/.arduino15/packages/arduino/tools/openocd/0.11.0-arduino2/bin/openocd -d2 -s ~/.arduino15/packages/arduino/tools/openocd/0.11.0-arduino2/share/openocd/scripts/ -f interface/stlink.cfg -f target/stm32h7x_dual_bank.cfg -c "telnet_port disabled; init; reset init; halt; adapter speed 10000; program $compiled_sketch_elf; reset run; shutdown"
```

closed 2022-05-11 by facchinm

GunnerDx88, 2022-05-13:
Thank you for your answer.
Just a quick question : When you say 'commandline' do you mean the command to add in the Config options window ?