arduino/ArduinoCore-mbed #479: Debugging Nicla Vision

simengangstad, 2022-05-11:
Hello,

Is there functionality for debugging the Nicla Vision over USB as one can do with the Nicla Sense Me? I see that the Nicla Sense Me has an ATSAM for providing debug functionality over USB, but as far I can see in the data sheet, this isn't the case for the Nicla Vision. If JTAG is the only option, which connector would one use which would fit on the pads on the board without soldering? The data sheet mentions that a 50 mil header can be used without soldering, but I can't seem to find such a connector.

giulcioffi, 2022-05-16:
Hi @simengangstad, as you said Nicla Sense ME can be debugged over USB thanks to the samd11 onboard, while you cannot do the same with Nicla Vision. However a debug interface is available of Nicla Vision fingers, which are accessible with 2 rows 1.27mm male headers as shown in these pictures.
[image]
[image]

closed 2022-05-16 by simengangstad

manchoz, 2022-11-17:
For future reference:

[How to debug sketches on Nicla Sense ME](https://forum.arduino.cc/t/how-to-debug-sketches-on-nicla-sense-me/1039133)