arduino/ArduinoCore-mbed #483: Can the Portenta full pinout .pdf be available as a .csv file?

hpssjellis, 2022-05-24:
**Can the Portenta full pinout .pdf be available as a .csv file?**

The full pinout .pdf is available at this link [here](https://content.arduino.cc/assets/Pinout-PortentaH7_latest.pdf), which is very useful, but a .csv file equivalent would be able to be sorted by column.

A column sort would make things easier when working with the 160 pins of the High Density connectors. 

A closed pin names issue is [here](https://github.com/arduino/ArduinoCore-mbed/issues/74) and an example sketch that generates pin names is [here](https://github.com/hpssjellis/portenta-pro-community-solutions/tree/main/examples/dot6-portenta-advanced/dot64-pin-names) 

[Here](https://github.com/hpssjellis/my-examples-for-the-arduino-portentaH7/blob/master/PINS/shared-best-so-far-csv-portenta-pins.csv) is my attempt at a .csv file that connects pin Names to pin Numbers and tries to add HD connector names. It is missing all HD connector pins that don't use standard pin names or connect to GND or VIN. As you can see it is not at all complete

hpssjellis, 2023-01-04:
This issue can close. I guess I can make my own CSV file if I really want one.

closed 2023-01-04 by hpssjellis

AleSanna, 2023-03-08:
Hi @hpssjellis, can you share the .csv file you made with the community (if possible)? Thanks

hpssjellis, 2023-03-08:
@AleSanna 

The reason I wanted an Arduino .CSV file is mine had so many inconsistencies. Here is my folder of attempts, which is old and probably very incorrect.. Please post a PR if you can make any improvements.   

[https://github.com/hpssjellis/my-examples-for-the-arduino-portentaH7/tree/master/PINS](https://github.com/hpssjellis/my-examples-for-the-arduino-portentaH7/tree/master/PINS)