arduino/ArduinoCore-mbed #489: QSPIFBlockDevice.h: No such file or directory (re Raspberry Pi Pico)

Gerriko, 2022-05-27:
Using mbed_rp2040/3.1.1

Board Variant: Raspberry Pi Pico

Have included the following libraries in my test example code (based off the AccessFlashAsUSBDisk example):
```
#include "PluggableUSBMSD.h"
#include "QSPIFBlockDevice.h"
#include "MBRBlockDevice.h"
#include "FATFileSystem.h"

```
When compiling I get the following response:

```
Alternatives for QSPIFBlockDevice.h: []
ResolveLibrary(QSPIFBlockDevice.h)
  -> candidates: []
AccessFlashAsUSBDisk:10:10: fatal error: QSPIFBlockDevice.h: No such file or directory
 #include "QSPIFBlockDevice.h"
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

Searching through the previous issues, I assume the fix will be similar to the one given in @[facchinm](https://github.com/facchinm) response & clarifications. Issue number: https://github.com/arduino/ArduinoCore-mbed/issues/377

If so, please could someone confirm, especially with regard to macros and pin numbers. Maybe some sort of guidance/instruction could be added to documentation and/or in the examples.

Gerriko, 2022-05-30:
@facchinm Thanks, I will try this out. BTW, I did wonder why I hadn't noticed this library before for Pico so I checked the library.properties file and noticed that the architectures attribute is missing mbed_rp2040 (it has architectures=mbed,mbed_nano,mbed_nicla). Also within the mbed_rp2040/3.1.1/libraries folder there is the USBMSD folder, which where I got the example from.... based on your comment, maybe this needs amending or removing.

facchinm, 2022-05-30:
It may be worth amending USBMSD examples (since the library is perfectly supported on RP2040 "plain"). MLC library only applies to boards with ST 6 axis accelerometer onboard, so the fact that it's missing from `mbed_rp2040` is correct.

Gerriko, 2022-05-31:
This works great thanks. Here's a simple example - not sure if `MassStorage.process();` is really required... I noticed it's used in MbedOS examples:

```
#include "PluggableUSBMSD.h"
#include "FlashIAPBlockDevice.h"

static FlashIAPBlockDevice bd(XIP_BASE + 0x100000, 0x100000);
USBMSD MassStorage(&bd);

static FILE *f = nullptr;
// Note that file is written to the actual root directory as created when USBMSD is mounted
// File is not stored within a folder called root.
// File is only found when you remove usb and replace again
const char *fname = "/root/myfile.txt";

void USBMSD::begin()
{
  int err = getFileSystem().mount(&bd);
  if (err) {
   //String FN = getFileSystem().getName();
   Serial.println(" filesystem mount failed\ntry to reformat device...");
    err = getFileSystem().reformat(&bd);
  }
  if (err) {
    Serial.println("Error: Unable to format/mount the device.");
    while(1);
  }
}

mbed::FATFileSystem &USBMSD::getFileSystem()
{
  static mbed::FATFileSystem fs("root");
  return fs;
}

void writeContents() {
    f = fopen(fname, "w+");
    if (f != nullptr) {
      fprintf(f, "Hello World\n");
      fflush(f);      
      fclose(f);
      Serial.println("Written to File");
    }
    else {
      Serial.println("File not found");
    }  
}

void readContents() {
    f = fopen(fname, "r");
    if (f != nullptr) {
      fclose(f);
      Serial.println("File found");
    }
    else {
      Serial.println("File not found");
    }  
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  MassStorage.begin();
  // If you do not want to wait for Serial Monitor to load then use a long delay here.
  // Found a delay helps if you want to capture the initial serial output.
  //while(!Serial) {;;}
  delay(1000);
  Serial.println("MassStorage mounted");
  writeContents();
  //readContents();

}

void loop() {
  // put your main code here, to run repeatedly:
  MassStorage.process();
}
```