arduino/ArduinoCore-mbed #490: SoftwareSerial for Portenta Machine control

KeerBot, 2022-05-29:
Hi, 

I tried to use SoftwareSerial.h with no success  - and found out that there's no option 
I'm using Portenta machine control (https://www.arduino.cc/pro/hardware/product/portenta-machine-control)

Any solution ? 

I'm trying to get serial data via RS232 from the computer and out put to a client machine also via RS232

hpssjellis, 2022-05-29:
@KeerBot I don't have the Machine Control board but this might help.

[Feature Request: SoftSerial For the Portenta small form factor](https://github.com/arduino/ArduinoCore-mbed/issues/364)

and [here](https://github.com/hpssjellis/portenta-pro-community-solutions/tree/main/examples/dot1-portentaH7-examples/dot19-cores-uart-serial-com) are my examples of using the advice from Arduino

MDBInd, 2022-09-25:
@KeerBot Did you have any luck with this? I am trying something similar with the Portenta Machine Control and its not working at the moment.
I did try and follow @hpssjellis post but without direct access to the PC_6 & PC_7 pins on the Portenta H7 I cannot get his example to work.

hpssjellis, 2022-09-25:
@MDBInd @KeerBot I feel your pain. Portenta Pro not supporting softserial means that hundreds of older sketches that easily connect peripherals can not be used with the new Arduino boards. Not sure about a solution but perhaps @facchinm can add his opinion. I don't own a Portenta Machine Control board but I feel that some of the 8 serial (UART) communication capable pins should be accessible.

facchinm, 2022-09-29:
The Portenta Machine Control I/O pins are all intended to run at higher voltage than usual equipment (hence the clear markings for their functions). Changing their intended function will likely not work or damage the device connected to them.

To talk with an RS232 device the pins marked as COMM PROTOCOLS exposes an RS485/RS232 port, so I think that the request is to connect **another** RS232 device on some other pins.

If it's the case, please reopen another issue and clearly specify the needed usecase :slightly_smiling_face:

closed 2022-09-29 by facchinm