arduino/ArduinoCore-mbed #498: Compatability with NINA-B306-01B

TobiasRoeddiger, 2022-06-11:
Hello :)

I have based a custom board on the Nano 33 BLE. Because the NINA-B306-00B is sold out, I would like to use the NINA-B306-01B (which is identical and only has an external crystal added to it). I have soldered it onto my PCB and flashed the bootloader.bin frome [here](https://github.com/arduino/ArduinoCore-mbed/tree/master/bootloaders/nano33ble) onto it via SWD using a jlink. I can see the device showing up in the Arduino IDE when I put it in DFU mode (press RESET twice) and also loaded sketches onto it in that state. However, it appears that something is going wrong as after flashing, the device does not show up on my list of USB devices and also it appears that the sketch is not running correctly on the Arduino (e.g., a simple LED blinky example does not work). 
I can not flash the device outside DFU mode as it is not showing up in the list of devices.

From my understanding, the only difference between the two modules is the 32.768 kHz crystal. Is that causing a breaking change in the software? I was thinking that the missing crystal will only cause some timing issues or so? Do I need a different bootloader to account for the different crystal (e.g., to avoid having bad Arduino delay() accuracy).

So it appears that some additional changes are required to make the the NINA-B306-01B compatible? Any help would be greatly appreciated. Thank you very much!

Thanks,
Tobias

facchinm, 2022-06-13:
Hi Tobias,
as you correctly found out, the issue is indeed due to the 32K oscillator. Since the -00B version had the crystal mounted, we didn't change the `MBED_CONF_NORDIC_NRF_LF_CLOCK_SRC` macro so it's still the default (`NRF_LF_SRC_XTAL`).
The startup code then tries to configure that clock here https://github.com/arduino/mbed-os/blob/extrapatches-6.15.1/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/TARGET_MCU_NRF52840/device/system_nrf52840.c#L186-L194 but obviously fails and never leaves the `while` loop.

The best solution is to modify the config file for the Nano 33 BLE (https://github.com/arduino/ArduinoCore-mbed/blob/master/variants/ARDUINO_NANO33BLE/conf/mbed_app.json), adding `"target.lf_clock_src":"NRF_LF_SRC_RC"` line (or `NRF_LF_SRC_SYNTH`) , then recompile the library using `mbed-os-to-arduino` script.

Let me know if it works for you! We may end up providing this "fix" in the official distribution soon but in the meantime doing it manually is the only way.

rylox, 2022-06-17:
Hey @facchinm,
I'm trying to implement this fix but I'm having some issues. Where is the script generated bootloader file copied to? I'm running it on debian and it prints out:  
`Setting up Mbed Application...[mbed] Working path "/tmp/mbed-os-program" (program)`
`[mbed] ARDUINO_NANO33BLE now set as default target in program "mbed-os-program"`
`[mbed] Working path "/tmp/mbed-os-program" (program)`
`[mbed] GCC_ARM now set as default toolchain in program "mbed-os-program"`
` done.`

Switching to `/tmp/mbed-os-program/BUILD/ARDUINO_NANO33BLE/GCC_ARM/` I have 3 flashable files `mbed-os-program.hex`, `mbed-os-program_application.hex` and `mbed-os-program_update.bin` none of which work. The mc won't get recognised after flashing them. The script finished successfully without any errors.
 
 Cheers
 Leon