arduino/ArduinoCore-mbed #504: Feature Request: PortentaH7 Camera working using M4

hpssjellis, 2022-06-20:
@facchinm (I wouldn't bug you if it wasn't important!)

Someone on the [edgeImpulse.com forum](https://forum.edgeimpulse.com/t/object-detection-on-m4-core/4778) is trying to get a machine learning vision model working on the M4 core. When I try even the simple Arduino [camera raw bytes](https://github.com/arduino/ArduinoCore-mbed/blob/master/libraries/Camera/examples/CameraCaptureRawBytes/CameraCaptureRawBytes.ino) example I get the same results as a complex sketch:

```
C:\Arduino-1.8.19\portable\packages\arduino\hardware\mbed_portenta\3.1.1\libraries\Camera\src\camera.cpp: In function 'void HAL_TIM_PWM_MspInit(TIM_HandleTypeDef*)':
C:\Arduino-1.8.19\portable\packages\arduino\hardware\mbed_portenta\3.1.1\libraries\Camera\src\camera.cpp:120:27: error: 'DCMI_TIM' was not declared in this scope
     if (htim->Instance == DCMI_TIM) {

```

Is this something that might have an easy fix or is it more complex than that?

facchinm, 2022-06-21:
Hi Jeremy,
this is due to [this](https://github.com/arduino/ArduinoCore-mbed/blob/master/libraries/Camera/src/camera.cpp#L31) and [this](https://github.com/arduino/ArduinoCore-mbed/blob/master/libraries/Camera/src/camera.cpp#L68) lines, that should read `#if defined(PORTENTA_H7_M7) || defined(PORTENTA_H7_M4)` to apply to M4 core too.
Let me know if changing it makes the camera work on the M4, because there may be other issues with memory zones and DMA. If it works, I'll be glad to merge the patch immadiately

closed 2022-06-21 by facchinm

hpssjellis, 2022-06-21:
Thanks @facchinm . I spent the whole day on it, summarized in the README.md file [here](https://github.com/hpssjellis/my-examples-for-the-arduino-portentaH7/tree/master/vision-shield/camera-m4)

Basically ```Camera::grabFrame(``` needs ```SCB_InvalidateDCache_by_Addr(```, but I can't seem to get it included for the PortentaH7 M4 core, it runs fine on the M7 Core. 

The file ```cachel1_armv7.h``` that has ```SCB_InvalidateDCache_by_Addr(``` is [here](https://github.com/arduino/ArduinoCore-mbed/blob/master/cores/arduino/mbed/cmsis/CMSIS_5/CMSIS/TARGET_CORTEX_M/Include/cachel1_armv7.h)

```\cores\arduino\mbed\cmsis\CMSIS_5\CMSIS\TARGET_CORTEX_M\Include\cachel1_armv7.h```

I am testing using the Arduino example [CameraMotionDetect](https://github.com/arduino/ArduinoCore-mbed/blob/master/libraries/Camera/examples/CameraMotionDetect/CameraMotionDetect.ino) I am getting the same error even if I switch to SDRAM or the HEAP for the camera buffer, those examples are [here](https://github.com/hpssjellis/portenta-pro-community-solutions/tree/main/examples/dot3-portenta-vision-shields/dot35-camera-and-oled)