arduino/ArduinoCore-mbed #506: not working with IDE 2.0.0 rc

basikv, 2022-06-21:
i am wondering why the below code is not working with IDE, where i got the output in the older version 1.8.9 without adding any libraries. can someone suggest some changes?
#define IN_PIN A0  
#define WINDOW_SIZE 20 
int INDEX = 0;
int value = 0;
int sum = 0;
int reading[WINDOW_SIZE];
int averaged = 0;

 void setup() {
  
  pinMode(IN_PIN, INPUT);
  analogReadResolution(16);
  Serial.begin(115200);    
}

  void loop() {
  sum = sum - reading[INDEX];       // Remove the oldest entry from the sum
  value = analogRead(IN_PIN);        // Read the next sensor value
  reading[INDEX] = value;           // Add the newest reading to the window
  sum = sum + value;                 // Add the newest reading to the sum
  INDEX = (INDEX + 1) % WINDOW_SIZE;   // Increment the index, and wrap to 0 if it exceeds the window size 
  averaged = sum / WINDOW_SIZE; // Divide the sum of the window by the window size for the result

  Serial.print(value);
  Serial.print(",");
  Serial.println(averaged);
  
  delay(1);
  }

per1234, 2022-06-21:
I am already providing support for this issue here:

https://forum.arduino.cc/t/not-working-on-ide-2-0-0-rc-version/1005149

I think that will be the best place for the current phase of the investigation. If a defect in some of Arduino's software is eventually identified, I will be make sure that a formal report is submitted to the developers.

So for now, please close this issue @basikv

basikv, 2022-06-21:
C:\Users\stutt\AppData\Local\Temp\arduino-sketch-61AD9CC8915FA7E426C588871BBB32BB\sketch\sketch_jun07a.ino.cpp.o: In function `loop':
C:\Users\stutt\Documents\Arduino\sketch_jun07a/sketch_jun07a.ino:18: undefined reference to `analogRead'
C:\Users\stutt\Documents\Arduino\sketch_jun07a/sketch_jun07a.ino:28: undefined reference to `delay'
C:\Users\stutt\AppData\Local\Temp\arduino-sketch-61AD9CC8915FA7E426C588871BBB32BB\sketch\sketch_jun07a.ino.cpp.o: In function `setup':
C:\Users\stutt\Documents\Arduino\sketch_jun07a/sketch_jun07a.ino:11: undefined reference to `pinMode(unsigned char, PinMode)'
C:\Users\stutt\Documents\Arduino\sketch_jun07a/sketch_jun07a.ino:12: undefined reference to `analogReadResolution'
C:\Users\stutt\AppData\Local\Temp\arduino-sketch-61AD9CC8915FA7E426C588871BBB32BB\core\variant.cpp.o: In function `initVariant':
C:\Users\stutt\AppData\Local\Arduino15\packages\arduino\hardware\mbed_portenta\3.1.1\variants\PORTENTA_H7_M7/variant.cpp:241: undefined reference to `pinMode(PinName, PinMode)'
C:\Users\stutt\AppData\Local\Arduino15\packages\arduino\hardware\mbed_portenta\3.1.1\variants\PORTENTA_H7_M7/variant.cpp:242: undefined reference to `digitalWrite(PinName, PinStatus)'
C:\Users\stutt\AppData\Local\Temp\arduino-sketch-61AD9CC8915FA7E426C588871BBB32BB/..\arduino-core-cache\core_arduino_mbed_portenta_envie_m7_split_50_50,security_none_d944dcf72ce0f10a56642b13c6cd2075.a(Serial.cpp.o): In function `arduino::UART::operator bool()':
C:\Users\stutt\AppData\Local\Arduino15\packages\arduino\hardware\mbed_portenta\3.1.1\cores\arduino\USB/PluggableUSBSerial.h:272: undefined reference to `delay'
C:\Users\stutt\AppData\Local\Temp\arduino-sketch-61AD9CC8915FA7E426C588871BBB32BB/..\arduino-core-cache\core_arduino_mbed_portenta_envie_m7_split_50_50,security_none_d944dcf72ce0f10a56642b13c6cd2075.a(USBSerial.cpp.o): In function `arduino::USBSerial::operator bool()':
C:\Users\stutt\AppData\Local\Arduino15\packages\arduino\hardware\mbed_portenta\3.1.1\cores\arduino\USB/PluggableUSBSerial.h:272: undefined reference to `delay'
C:\Users\stutt\AppData\Local\Temp\arduino-sketch-61AD9CC8915FA7E426C588871BBB32BB/..\arduino-core-cache\core_arduino_mbed_portenta_envie_m7_split_50_50,security_none_d944dcf72ce0f10a56642b13c6cd2075.a(USBSerial.cpp.o): In function `waitForPortClose()':
C:\Users\stutt\AppData\Local\Arduino15\packages\arduino\hardware\mbed_portenta\3.1.1\cores\arduino\USB/USBSerial.cpp:35: undefined reference to `millis'
C:\Users\stutt\AppData\Local\Arduino15\packages\arduino\hardware\mbed_portenta\3.1.1\cores\arduino\USB/USBSerial.cpp:39: undefined reference to `delay'
C:\Users\stutt\AppData\Local\Arduino15\packages\arduino\hardware\mbed_portenta\3.1.1\cores\arduino\USB/USBSerial.cpp:37: undefined reference to `millis'
C:\Users\stutt\AppData\Local\Temp\arduino-sketch-61AD9CC8915FA7E426C588871BBB32BB/..\arduino-core-cache\core_arduino_mbed_portenta_envie_m7_split_50_50,security_none_d944dcf72ce0f10a56642b13c6cd2075.a(main.cpp.o): In function `main':
C:\Users\stutt\AppData\Local\Arduino15\packages\arduino\hardware\mbed_portenta\3.1.1\cores\arduino/main.cpp:36: undefined reference to `init'
collect2.exe: error: ld returned 1 exit status

Compilation error: exit status 1
C:\Users\stutt\AppData\Local\Temp\arduino-sketch-61AD9CC8915FA7E426C588871BBB32BB\sketch\sketch_jun07a.ino.cpp.o: In function `loop':
C:\Users\stutt\Documents\Arduino\sketch_jun07a/sketch_jun07a.ino:18: undefined reference to `analogRead'
C:\Users\stutt\Documents\Arduino\sketch_jun07a/sketch_jun07a.ino:28: undefined reference to `delay'
C:\Users\stutt\AppData\Local\Temp\arduino-sketch-61AD9CC8915FA7E426C588871BBB32BB\sketch\sketch_jun07a.ino.cpp.o: In function `setup':
C:\Users\stutt\Documents\Arduino\sketch_jun07a/sketch_jun07a.ino:11: undefined reference to `pinMode(unsigned char, PinMode)'
C:\Users\stutt\Documents\Arduino\sketch_jun07a/sketch_jun07a.ino:12: undefined reference to `analogReadResolution'
C:\Users\stutt\AppData\Local\Temp\arduino-sketch-61AD9CC8915FA7E426C588871BBB32BB\core\variant.cpp.o: In function `initVariant':
C:\Users\stutt\AppData\Local\Arduino15\packages\arduino\hardware\mbed_portenta\3.1.1\variants\PORTENTA_H7_M7/variant.cpp:241: undefined reference to `pinMode(PinName, PinMode)'
C:\Users\stutt\AppData\Local\Arduino15\packages\arduino\hardware\mbed_portenta\3.1.1\variants\PORTENTA_H7_M7/variant.cpp:242: undefined reference to `digitalWrite(PinName, PinStatus)'
C:\Users\stutt\AppData\Local\Temp\arduino-sketch-61AD9CC8915FA7E426C588871BBB32BB/..\arduino-core-cache\core_arduino_mbed_portenta_envie_m7_split_50_50,security_none_d944dcf72ce0f10a56642b13c6cd2075.a(Serial.cpp.o): In function `arduino::UART::operator bool()':
C:\Users\stutt\AppData\Local\Arduino15\packages\arduino\hardware\mbed_portenta\3.1.1\cores\arduino\USB/PluggableUSBSerial.h:272: undefined reference to `delay'
C:\Users\stutt\AppData\Local\Temp\arduino-sketch-61AD9CC8915FA7E426C588871BBB32BB/..\arduino-core-cache\core_arduino_mbed_portenta_envie_m7_split_50_50,security_none_d944dcf72ce0f10a56642b13c6cd2075.a(USBSerial.cpp.o): In function `arduino::USBSerial::operator bool()':
C:\Users\stutt\AppData\Local\Arduino15\packages\arduino\hardware\mbed_portenta\3.1.1\cores\arduino\USB/PluggableUSBSerial.h:272: undefined reference to `delay'
C:\Users\stutt\AppData\Local\Temp\arduino-sketch-61AD9CC8915FA7E426C588871BBB32BB/..\arduino-core-cache\core_arduino_mbed_portenta_envie_m7_split_50_50,security_none_d944dcf72ce0f10a56642b13c6cd2075.a(USBSerial.cpp.o): [... cut ...]
C:\Users\stutt\AppData\Local\Arduino15\packages\arduino\hardware\mbed_portenta\3.1.1\cores\arduino\USB/USBSerial.cpp:37: undefined reference to `millis'
C:\Users\stutt\AppData\Local\Temp\arduino-sketch-61AD9CC8915FA7E426C588871BBB32BB/..\arduino-core-cache\core_arduino_mbed_portenta_envie_m7_split_50_50,security_none_d944dcf72ce0f10a56642b13c6cd2075.a(main.cpp.o): In function `main':
C:\Users\stutt\AppData\Local\Arduino15\packages\arduino\hardware\mbed_portenta\3.1.1\cores\arduino/main.cpp:36: undefined reference to `init'
collect2.exe: error: ld returned 1 exit status

Compilation error: exit status 1

:-this what its shows