arduino/ArduinoCore-mbed #508: Bootloader compilation and location

rylox, 2022-06-21:
I'm trying to implement [this fix](https://github.com/arduino/ArduinoCore-mbed/issues/498) but I'm having some issues. Where is the script generated bootloader file copied to? I'm running it on debian and it prints out:
Setting up Mbed Application...[mbed] Working path "/tmp/mbed-os-program" (program)
[mbed] ARDUINO_NANO33BLE now set as default target in program "mbed-os-program"
[mbed] Working path "/tmp/mbed-os-program" (program)
[mbed] GCC_ARM now set as default toolchain in program "mbed-os-program"
done.

Switching to /tmp/mbed-os-program/BUILD/ARDUINO_NANO33BLE/GCC_ARM/ I have 3 flashable files mbed-os-program.hex, mbed-os-program_application.hex and mbed-os-program_update.bin none of which work. The mc won't get recognised after flashing them. The script finished successfully without any errors. I also tested all files in DFU mode.

Cheers
Leon

Xraydylan, 2022-06-21:
I have encountered a similar problem...

facchinm, 2022-06-22:
Hi guys, once  `mbed-os-to-arduino` script executes the correct files (`libmbed.a` , all flags and includes, the linker script) gets copied to `variant/VARIANT_NAME/`. My advice is to use that core as-is following https://github.com/arduino/ArduinoCore-mbed#installation , so you'll find it in a submenu, completely separated from the "normal" installation

[image]

All the other files from `/tmp/mbed-os-program/BUILD/ARDUINO_NANO33BLE/` are bogus and should not be flashed. For the bootloader you can safely use the hex from https://github.com/arduino/ArduinoCore-mbed/tree/master/bootloaders/nano33ble that's also compatible with 01B modules after applying this patch https://github.com/arduino/ArduinoCore-mbed/pull/509.

closed 2022-06-22 by facchinm