arduino/ArduinoCore-mbed #522: Custom Variant with RP2040

sabas1080, 2022-07-19:
Hello
I am creating a custom variant for my devboard, the first start is correct but when restarting the program does not start, with the core of https://github.com/earlephilhower/arduino-pico it works correctly

Memory flash: https://www.mouser.mx/ProductDetail/Winbond/W25Q16JVUUIQ-TR?qs=qSfuJ%252Bfl%2Fd7XfZZWV6GUpQ%3D%3D

Thanks

facchinm, 2022-07-20:
Hi @sabas1080 ,
if it starts correctly on the first run it means that the [second stage bootloader](https://github.com/arduino/mbed-os/blob/extrapatches-6.15.1/targets/TARGET_RASPBERRYPI/TARGET_RP2040/TARGET_RASPBERRY_PI_PICO/bs2_default_padded_checksummed.S) is fine (it would have been the culprit since it's tied to the flash chip).
What do you mean with "restarting the program"? Just press the reset button?

sabas1080, 2022-07-27:
@facchinm Do you have any possible theory? Thanks

sabas1080, 2022-08-04:
@facchinm What is the correct procedure/steps to add a variant of RP2040? I think that's where the problem

sabas1080, 2022-08-04:
@facchinm I have found the error in the crystal

sabas1080, 2022-08-04:
https://github.com/raspberrypi/pico-sdk/pull/869

closed 2023-02-01 by sabas1080