arduino/ArduinoCore-mbed #529: Portenta X8 examples: QSPIFBlockDevice not found

ClemensAtElektor, 2022-08-01:
Hi,

Tried to run WiFiFirmwareUpdater & WiFiWebClient examples from Portenta X8 boards package v3.2.0 inside Arduino IDE 1.8.19 on my Portenta X8. Both fail with similar compilation error:

`ResolveLibrary(QSPIFBlockDevice.h)WiFiFirmwareUpdater:1:10: fatal error: QSPIFBlockDevice.h: No such file or directory`

I saw other issues related to QSPIFBlockDevice here (#489 , #377) but I am not sure if I can apply the proposed solutions. How can I get this to compile, please?

facchinm, 2022-08-01:
Hi @ClemensAtElektor ,
the X8 cannot use WiFi libraries directly from the Arduino side of things (you can however interfae with the Linux side handling the network communication using the [RPC examples](https://docs.arduino.cc/tutorials/portenta-x8/python-arduino-data-exchange) ).
@sebromero should we be a bit more explicit with the error when the library being included is not compatible with X8?

closed 2022-08-01 by facchinm

ClemensAtElektor, 2022-08-01:
Thanks for the quick answer. However, you closed this issue, but are you aware that in the Arduino IDE these examples are listed as "Examples for Arduino Portenta X8"? (see attached screenshot) As such, one would expect them to compile without errors.

[image]

facchinm, 2022-08-01:
@ClemensAtElektor you are right, this is a big shortcoming of the IDE; it shows all the libraries in a core as `Example for $boardname` . The same happens on AVR core for boards without USB support.
@per1234 is there any plan to fix this in IDE 2? Maybe adding some extra info in a `library.properties` field?

per1234, 2022-08-07:
>is there any plan to fix this in IDE 2?

There are a couple of proposals for doing such a thing at the library or sketch level (keep in mind that, even if a library is compatible with a board, a specific example sketch may not be). See issues 7144 and 5506 in the Tooling Team's internal issue tracker (unfortunately these were created before we had achieved proper integration with GitHub and we haven't gotten around to publishing them yet). But the most detailed proposal is the document you wrote in the "**Arduino Tooling**" confluence space.

Those proposals are likely to take some time to implement since they encompass work in the Arduino CLI code base, the [relevant specification](https://arduino.github.io/arduino-cli/latest/library-specification), [Arduino IDE](https://github.com/arduino/arduino-ide), [Arduino Web Editor](https://create.arduino.cc/editor), the relevant [boards platforms](https://arduino.github.io/arduino-cli/latest/platform-specification/), before benefiting users. After that, there will be follow up work in [Arduino Lint](https://github.com/arduino/arduino-lint), [Arduino Library Reference](https://www.arduino.cc/reference/en/libraries/), and [Arduino Library Manager](https://docs.arduino.cc/software/ide-v2/tutorials/ide-v2-installing-a-library).

I have submitted a proposal for an improvement that can be made to this user experience with only a trivial change to the Arduino IDE code: https://github.com/arduino/arduino-ide/issues/1304.

marimelhanafy, 2023-09-18:
> Hi @ClemensAtElektor , the X8 cannot use WiFi libraries directly from the Arduino side of things (you can however interfae with the Linux side handling the network communication using the [RPC [...]

What if I want to receive data from WiFi? I have a testbed that consists of 4 ESP-WROOM-32 units, each connected to different kinds of sensors.  I want to send the data from all of them to the Portenta X8 simultaneously. I have already completed the ESPs part, as I connected all of them to the WiFi and sent the data using the TCP protocol. I used these headers (#include <WiFi.h> / #include <AsyncTCP.h> / #include <ESPAsyncWebSrv.h>).
My question is: how should I receive the data on the Portenta X8? I searched but did not find any related header for it.  So, could you please let me know what the possible solutions are for this?