arduino/ArduinoCore-mbed #541: nRF52 OTA DFU

baconcheese113, 2022-08-24:
The nRF Connect SDK for the nRF52 in the Nano 33 BLE provides an ability to do OTA firmware updates https://infocenter.nordicsemi.com/index.jsp?topic=%2Fug_nrfconnect_ble%2FUG%2FnRF_Connect_BLE%2FnRF_Connect_DFU.html

It doesn't appear as though there are any solutions currently to get this functionality using the Arduino/mbed platform core seeing as ArduinoOTA only supports the nRF51 (see [this JAndrassy/ArduinoOTA issue](https://github.com/JAndrassy/ArduinoOTA/issues/49#issuecomment-1213042123) and [this sandeepmistry/arduino-nRF5 issue](https://github.com/sandeepmistry/arduino-nRF5/issues/395#issuecomment-643711567))

My question is about what would be required to get that functionality with the Arduino/mbed platform core? 

Also, for my understanding, do Arduino cores use vendor SDK's? Is there an article or documentation that would explain that linkage?

facchinm, 2022-08-25:
Hi @baconcheese113 ,
this core doesn't use Nordic Softdevices so OTA doesn't work out of the box.
However, we implemented a DFU protocol over ArduinoBLE for Nicla Sense (https://github.com/arduino/nicla-sense-me-fw/blob/main/Arduino_BHY2/src/DFUManager.cpp) and it can be ported to Nano33BLE quite easily :wink:

closed 2022-08-25 by facchinm

AccelElec, 2023-04-17:
Where did we land on this issue? The repo currently includes system_nrf52840.c and several other nRF52 device specific files which might imply that this has changed, yes?