arduino/ArduinoCore-mbed #542: mbed USBAudio breaks Arduino serial uploading

w9ran, 2022-08-28:
I'm interested in using the mbed USBAudio class to send audio from the Pico to a PC. I found some examples and made them work, but I think the author maybe did not properly handle the USB port, because after playing USB audio both ways the Pico always showed up as "mbed Audio" in Device Manager rather than as a serial port. I could compile new programs that didn't use the USBAudio class but was unable to upload them because at the end of the compile the Pico would appear as a new USB Storage Device and the following error message woult be displayed:

rp2040load 1.0.1 - compiled with go1.15.8
.ERROR: Could not open 'C:\Users\user\AppData\Local\Temp\arduino_build_664182/test_pico.ino.elf.uf2'

the selected serial port
does not exist or your board is not connected

There is indeed no uf2 file, but there is an elf file in the Pico build directory and I have a utility called elf2uf2 that I can use to make a uf2 file that I can then drop on the Pico USB storage device which will accept it and run the new code. As an example, I dropped a "blink" example this way and the Pico indeed starts blinking. But it's obviously a cumbersome process.

I even tried a brand new factory-fresh Pico board but when I plugged it in it came up as a USB memory device without holding down the boot sel button instead of as a serial device. So of course the serial upload could not work when I tried sending the blink program to it because there's no serial port available. Connecting the blank Pico to USB while holding down the boot sel button makes it appear as a USB storage device as it should but for some reason plugging it in without pressing boot sel does the same thing now, instead of showing up as a serial device. Prior to these experiments the expected functionality worked fine on this PC.

I'll paste the verbose errors below. It can't find COM9 because there is nothing at COM9 in device manager, although a different Pico board did use that port.

I first thought the USBAudio experiments had locked up the USB port but uploading is impossible after uninstalling it and even rebooting the PC. Something seems to be wrong with the mbed Pico support environment and I'd certainly appreciate any suggestions about how to fix it.

Bob

Sketch uses 78604 bytes (0%) of program storage space. Maximum is 16777216 bytes.
Global variables use 42788 bytes (15%) of dynamic memory, leaving 227548 bytes for local variables. Maximum is 270336 bytes.
Forcing reset using 1200bps open/close on port COM9
processing.app.debug.RunnerException
at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:152)
at cc.arduino.UploaderUtils.upload(UploaderUtils.java:77)
at processing.app.SketchController.upload(SketchController.java:732)
at processing.app.SketchController.exportApplet(SketchController.java:703)
at processing.app.Editor$UploadHandler.run(Editor.java:2070)
at java.lang.Thread.run(Thread.java:748)
Caused by: processing.app.SerialException: Error touching serial port 'COM9'.
at processing.app.Serial.touchForCDCReset(Serial.java:107)
at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:136)
... 5 more
Caused by: jssc.SerialPortException: Port name - COM9; Method name - openPort(); Exception type - Port not found.
at jssc.SerialPort.openPort(SerialPort.java:167)
at processing.app.Serial.touchForCDCReset(Serial.java:101)
... 6 more

facchinm, 2022-08-29:
Hi Bob,
you are experiencing a couple of different issues, let's try to split them.

1 - The board looks "dead" and keeps showing up as mass storage
Uploading a blink (don't worry about the missing serial port, the IDE should upload even if it's only in mass storage mode) should get you back the serial port and restore the board

2 - USBAudio
Since you are on windows, the process can be more cumbersome than with other OS due to drivers reenumeration. Some manual intervention might be needed.

3 - No serial port after uploading USBAudio sketch
Including directly "USBAudio.h" replaces the USB stack completely; the right way to keep the serial port is including "PluggableUSBAudio.h", then the APIs are the same.

I'm attaching a sketch that outputs a ~1300hz sine wave from the Pico, hope it helps fixing your issue

```
#include "PluggableUSBAudio.h"

USBAudio audio(true, 44100, 2, 44100, 2);

static uint8_t buf[128];

void setup() {
  for (int i = 0; i<sizeof(buf); i++) {
    buf[i] = 128 * sin(i);
  }
}

void loop() {
  audio.write(buf, sizeof(buf));
}
```

closed 2022-08-29 by facchinm

w9ran, 2022-09-03:
Thanks for your help. 

One more thing - I've looked all over - can you tell me where to find PluggableUSBAudio.h?

Bob

facchinm, 2023-05-08:
Ooops, the library is not being included in the core build for RP2040, but you can recover it from here https://github.com/arduino/ArduinoCore-mbed/tree/main/libraries/USBAudio